ansys/pyaedt #2617: Hfss.export_results() docstring

dcrawforAtAnsys, 2023-03-10:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

[export_results()](https://aedt.docs.pyansys.com/release/0.6/API/_autosummary/pyaedt.hfss.Hfss.export_results.html) is inherited from Analysis and the docstring for many input paramters is difficult to understand and not applicable to HFSS.
- length: The meaning of this paramter is not clear from the documentation nor for HFSS users.

### Steps To Reproduce

This is a documentation issue.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

Not related to usage or execution.

dcrawforAtAnsys, 2023-03-25:
Documentation has been improved to mention applicability of the method for Q3D, Q2D, etc.

Samuelopez-ansys, 2023-03-28:
@dcrawforAtAnsys Did you improve the documentation? I don't see any modification regarding the applicability of the parameter 'length' for Q2D designs.

dcrawforAtAnsys, 2023-03-28:
@Samuelopez-ansys : It looks like the docstring was updated by @maxcapodi78 on 08-March-23.

closed 2023-03-28 by dcrawforAtAnsys