ansys/pyaedt #2618: Bug located in ...

AndyInHollis, 2023-03-10:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

IPC-2581 file from pyAEDT IPC Export process is missing layer-associations when opened in Cadence/Allegro.
While the Etch shapes resulting from the IPC2581.xml file export are present visually when Global Visibility is first enabled, and the Layer Stackup is present, however the actual copper Etch planes do not contain the Etch or Dielectric shapes necessary to have a functional BRD design file.  
It appears that the pyAEDT Export process lacks the associations between shape and layer within the IPC2581 File.
The script I wrote for V2023/R1 seems to process error-free in Spyder.

To see this, turn off Global Visibility and then turn on any Layer in Etch, and nothing is present.

My theory is that an association portion or tag in the XML file is missing.

### Steps To Reproduce

The process I am following is as follows:
1) Import an Allegro BRD file into V2023/R1 (latest).
[694-775-01_MLO_081622_FCS_Cutout_2023R1.zip](https://github.com/pyansys/pyaedt/files/10946152/694-775-01_MLO_081622_FCS_Cutout_2023R1.zip)

2) Cut out a section, which I have done. (Archive File of cutout is attached, full file is >250GB)
(Not I get an error message when trying to attach an aedtz file so I changed the extension to simple zip)
The eventual purpose is to modify in HFSS/3DLayout and then export using IPC2581 so I can import back into Allegro for our normal post-processing after Sim in HFSS is successful.
The Cutout is successful in 3D Layout.

3) Export to IPC2581 with pyAEDT (Exported file is attached)  (pyAEDT Script is attached)

4) Import into Allegro and view (resulting BRD file is attached).
[ipc2581 Three Files.zip](https://github.com/pyansys/pyaedt/files/10946211/ipc2581.Three.Files.zip)

Note, none of the file extensions I tried to upload were supported, so I Zipped the remaining three files into a single Zip archive.

However, upon import into Allegro, real layers do not exist.
My theory is that an association portion or tag in the XML file is missing.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

alabaster==0.7.13
arrow==1.2.3
astroid==2.14.2
asttokens==2.2.1
atomicwrites==1.4.1
attrs==22.2.0
autopep8==1.6.0
Babel==2.12.1
backcall==0.2.0
bcrypt==4.0.1
beautifulsoup4==4.11.2
binaryornot==0.4.4
black==23.1.0
bleach==6.0.0
certifi==2022.12.7
cffi==1.15.1
chardet==5.1.0
charset-normalizer==3.0.1
click==8.1.3
cloudpickle==2.2.1
clr-loader==0.2.5
colorama==0.4.6
comm==0.1.2
contourpy==1.0.7
cookiecutter==2.1.1
cryptography==39.0.2
cycler==0.11.0
debugpy==1.6.6
decorator==5.1.1
defusedxml==0.7.1
diff-match-patch==20200713
dill==0.3.6
docstring-to-markdown==0.11
docutils==0.19
entrypoints==0.4
executing==1.2.0
fastjsonschema==2.16.3
flake8==6.0.0
fonttools==4.38.0
idna==3.4
imageio==2.26.0
imagesize==1.4.1
importlib-metadata==6.0.0
inflection==0.5.1
intervaltree==3.1.0
ipykernel==6.21.2
ipython==8.11.0
ipython-genutils==0.2.0
isort==5.12.0
jaraco.classes==3.2.3
jedi==0.18.2
jellyfish==0.9.0
Jinja2==3.1.2
jinja2-time==0.2.0
jsonschema==4.17.3
jupyter_client==7.4.9
jupyter_core==5.2.0
jupyterlab-pygments==0.2.2
keyring==23.13.1
kiwisolver==1.4.4
lazy-object-proxy==1.9.0
MarkupSafe==2.1.2
matplotlib==3.7.0
matplotlib-inline==0.1.6
mccabe==0.7.0
mistune==2.0.5
more-itertools==9.1.0
mypy-extensions==1.0.0
nbclient==0.7.2
nbconvert==7.2.9
nbformat==5.7.3
nest-asyncio==1.5.6
numpy==1.24.2
numpydoc==1.5.0
packaging==23.0
pandas==1.5.3
pandocfilters==1.5.0
paramiko==3.0.0
parso==0.8.3
pathspec==0.11.0
pexpect==4.8.0
pickleshare==0.7.5
Pillow==9.4.0
platformdirs==3.0.0
pluggy==1.0.0
plumbum==1.8.1
pooch==1.7.0
prompt-toolkit==3.0.38
psutil==5.9.4
ptyprocess==0.7.0
pure-eval==0.2.2
pyaedt==0.6.48
pycodestyle==2.10.0
pycparser==2.21
pydocstyle==6.2.3
pyflakes==3.0.1
Pygments==2.14.0
pylint==2.16.3
pylint-venv==3.0.1
pyls-spyder==0.4.0
PyNaCl==1.5.0
pyparsing==3.0.9
PyQt5==5.15.9
PyQt5-Qt5==5.15.2
PyQt5-sip==12.11.1
PyQtWebEngine==5.15.6
PyQtWebEngine-Qt5==5.15.2
pyrsistent==0.19.3
python-dateutil==2.8.2
python-lsp-black==1.2.1
python-lsp-jsonrpc==1.0.0
python-lsp-server==1.7.1
python-slugify==8.0.1
pythonnet==3.0.1
pytoolconfig==1.2.5
pytz==2022.7.1
pyvista==0.38.3
pywin32==305
pywin32-ctypes==0.2.0
PyYAML==6.0
pyzmq==25.0.0
QDarkStyle==3.0.3
qstylizer==0.2.2
QtAwesome==1.2.2
qtconsole==5.4.0
QtPy==2.3.0
requests==2.28.2
rope==1.7.0
rpyc==5.3.0
Rtree==1.0.1
scooby==0.7.1
six==1.16.0
snowballstemmer==2.2.0
sortedcontainers==2.4.0
soupsieve==2.4
Sphinx==6.1.3
sphinxcontrib-applehelp==1.0.4
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==2.0.1
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.5
spyder==5.4.2
spyder-kernels==2.4.2
stack-data==0.6.2
text-unidecode==1.3
textdistance==4.5.0
three-merge==0.1.1
tinycss2==1.2.1
toml==0.10.2
tomli==2.0.1
tomlkit==0.11.6
tornado==6.2
traitlets==5.9.0
typing_extensions==4.5.0
ujson==5.7.0
urllib3==1.26.14
vtk==9.2.6
watchdog==2.3.1
wcwidth==0.2.6
webencodings==0.5.1
whatthepatch==1.0.4
wrapt==1.15.0
yapf==0.32.0
zipp==3.15.0

AndyInHollis, 2023-03-13:
Thanks Maxime, title change is helpful

svandenb-dev, 2023-03-16:
thanks for your feedback @AndyInHollis we will have a look

ring630, 2023-08-06:
@AndyInHollis  @svandenb-dev, primitives are placed on manufacturing layers. They should be on stackup layers.

AndyInHollis, 2023-08-07:
> CAUTION - EXTERNAL EMAIL: Do not open attachments or links unless you recognize the sender and the content is safe.
>
> @AndyInHollis [github.com]<https://urldefense.com/v3/__https:/github.com/AndyInHollis__;!!GenOTVeOfQ!QU8WaZRBOHEDSZQ7m-p48hxidbP3vcOkmBI-tzBfn0FrxFjXob4nYFYOVfMtGSNaHel7J-KtWUOxi_6fFweKBvEbC5DIixQ$> @svandenb-dev [github.com]<https://urldefense.com/v3/__https:/github.com/svandenb-dev__;!!GenOTVeOfQ!QU8WaZRBOHEDSZQ7m-p48hxidbP3vcOkmBI-tzBfn0FrxFjXob4nYFYOVfMtGSNaHel7J-KtWUOxi_6fFweKBvEbfCXkBvo$>, primitives are placed on manufacturing layers. They should be on stackup layers.
>
>
>
> Please note that this message may contain confidential information. If you have received this message by mistake, please inform the sender of the mistake, then delete the message from your system without making, distributing or retaining any copies of it. Although we believe that the message and any attachments are free from viruses and other errors that might affect the computer or IT system where it is received and read, the recipient opens the message at his or her own risk. We assume no responsibility for any loss or damage arising from the receipt or use of this message. When we process personal data relating to physical persons, such processing will meet the requirements of applicable data protection legislation and will be in accordance with our Privacy Policy.<https://www.teradyne.com/privacy-statement>

Thanks for this response Hui, however I am afraid I do not understand your point.

Are you saying the Ansys pyAEDT translation processor is placing the primitives/copper-layer-objects on the wrong type of layer?

Please expand your point, I have not been able to overcome the issue I raised, and have been assuming this is an Allegro issue.

Being able to take full advantage of an HFSS/3D to Allegro Export would be quite valuable to my firm.

Right now, the pyAEDT Exported/translated file does not contain real layer information necessary to create object in Allegro that will allow the necessary Gerber export process.

ring630, 2023-08-08:
@AndyInHollis , that was my first thought. After a deeper look, it turns out not the case. All the layers are correctly placed in IPC.xml. There is a chance that Allegro doesn't read layer information correclty without a certain configure file.

I attached two ipc2581 files. The first one is pyaedt exported one. The second is the original file. SIwave can import both file without problem. 

[pyaedt_exported_ipc.xml.txt](https://github.com/ansys/pyaedt/files/12288573/pyaedt_exported_ipc.xml.txt)
[testcase9-RevC-Full.xml.txt](https://github.com/ansys/pyaedt/files/12288582/testcase9-RevC-Full.xml.txt)

I don't have Allegro license. I need your help to test them in Allegro.

AndyInHollis, 2023-08-09:
Excellent to see, I will try this by tomorrow afternoon (USA EST)

I did not think to use SIWave, I always use 3D-Layout, but I don’t think the different environments will respond differently.

I will report back.

@AndyInHollis [github.com]<https://urldefense.com/v3/__https:/github.com/AndyInHollis__;!!GenOTVeOfQ!WmZ2GXw8q_VfUAHcDkC0xp6YqgAlXe1WhX6JIXC6NmZzO1MpYpTYgmdECN6W5TETb1QTOOx3Dtl5gzNqgqwaEYaXLkqrbzs$> , that was my first thought. After a deeper look, it turns out not the case. All the layers are correctly placed in IPC.xml. There is a chance that Allegro doesn't read layer information correclty without a certain configure file.

pyaedt_exported_ipc.xml.txt [github.com]<https://urldefense.com/v3/__https:/github.com/ansys/pyaedt/files/12288573/pyaedt_exported_ipc.xml.txt__;!!GenOTVeOfQ!WmZ2GXw8q_VfUAHcDkC0xp6YqgAlXe1WhX6JIXC6NmZzO1MpYpTYgmdECN6W5TETb1QTOOx3Dtl5gzNqgqwaEYaX6N5rKS8$>
testcase9-RevC-Full.xml.txt [github.com]<https://urldefense.com/v3/__https:/github.com/ansys/pyaedt/files/12288582/testcase9-RevC-Full.xml.txt__;!!GenOTVeOfQ!WmZ2GXw8q_VfUAHcDkC0xp6YqgAlXe1WhX6JIXC6NmZzO1MpYpTYgmdECN6W5TETb1QTOOx3Dtl5gzNqgqwaEYaXNA0W_vM$>

ring630, 2023-09-27:
@AndyInHollis , is your problem solved?

ring630, 2023-10-02:
@AndyInHollis I have not hear from you, so I suppose your problem is solved.  I am closing this issue.

closed 2023-10-02 by ring630

AndyInHollis, 2023-10-02:
Thanks, I reported failure to the normal Ansys channel here at Teradyne.
But the issue appears to be with Cadence, so thanks for your note and case-closed

@AndyInHollis [github.com]<https://urldefense.com/v3/__https:/github.com/AndyInHollis__;!!GenOTVeOfQ!QJQotx22P8PklS9YVtTCbfbwmIiTLf8LFipaSyW2kcXPSz1CzcoQe6fjD7bkJuDxZm8xDrburA9mEju6JiIkrAVJ1k0hh8U$> I have not hear from you, so I suppose your problem is solved. I am closing this issue.