ansys/pyaedt #2630: Add Support for Cylindrical and Polar Coordinates in AEDT Materials

DaveTwyman, 2023-03-16:
### Description of the feature

Unless I've missed how to do this, we are unable to specify Cylindrical or Polar coordinates in Material properties for AEDT
Cylindrical Coordinates are really useful for setting magnetisation of Permanent magnets that are radially magnetised.

Polar coordinates are used less in materials but support could be added to the API whilst adding Cylindrical for complete coverage of native AEDT API.

Just for reference this is the screenshot of the UI
[image]

and the corresponding 

### Steps for implementing the feature

Expose functionality of the native IronPython 
oDefinitionManager.EditMaterial

to provide provide support for Cylindrical Coordinate System

POKowol, 2023-04-20:
I fully support the idea, it would be quite a useful function in modeling transducers with permanent magnets.

jmaravin, 2023-09-13:
Has there been any update on this? I fully support the idea as well.. it is very useful and needed

gmalinve, 2023-09-20:
Close this issue as duplicate of #3599 .

@jmaravin @DaveTwyman @POKowol hopefully someone will tackle this request soon.

DaveTwyman, 2023-09-28:
I'm going to reopen this issue, acknowledging that it is indeed a duplicate of https://github.com/ansys/pyaedt/issues/3599 

We have more of a history of who has requested it. More usefully I've self assigned the issue and will work on it next week

jmaravin, 2023-09-28:
Hi @DaveTwyman thanks for reopening.. looking forward to that new code.. 
Thanks for your help.

Regards, 
Jaime

gmalinve, 2023-09-28:
Hi @DaveTwyman @jmaravin ,

it's fine to re-open this issue but I'll close the one open by @jmaravin in order to avoid having duplicates.
@DaveTwyman if you create a PR please link it to this issue :) 

Thanks!

Giulia

Samuelopez-ansys, 2023-10-02:
@DaveTwyman @jmaravin 

Please check the linked pull request. I have implemented the coordinate system material property and I have also added the magnetic coercivity in the property material list, you will be able to modify each parameter:

```
import pyaedt

m = pyaedt.Maxwell2d(specified_version="2023.2")

new = m.materials.add_material("new")

m.materials.material_keys["new"].coordinate_system = "Cylindrical"

m.materials.material_keys["new"].magnetic_coercivity = ['5', '2', '1', '0']

m.release_desktop(False, False)

```

jmaravin, 2023-10-02:
Hi Samuel, thanks very much for your work on this. I just copied and pasted your code and I get the following "new" material. 
For some reason the fourth and fifth lines of the code are not read properly. See picture attached. 

[image]

Samuelopez-ansys, 2023-10-03:
Hi @jmaravin, as @DaveTwyman suggested, the branch is still not merged in the main. 

If you want to test it, you need to create a clone of the branch on your local computer: https://docs.github.com/en/repositories/creating-and-managing-repositories/cloning-a-repository

Once it is merged, it will be on PyAEDT main branch, but until we don't launch a new release, it will not be available trough 'pip', it is the open-source philosophy.

Regards

closed 2023-10-03 by Samuelopez-ansys

jmaravin, 2023-10-04:
Hi @Samuelopez-ansys , thanks again for having fixed this.. I find it really useful to what I do.. quick question that I have because I am not very familiar with how Github works. 
I´ve seen this new feature has been merged to the main branch as the Pull Request was acepted and closed. Does that mean I can use this new feature now?
Thanks again for your help and apologies as I am very new to Github. 
Jaime