ansys/pyaedt #2644: solve_in_batch method in Analysis.py has a bug in that it is not waiting for batch job to finish

vishak-venkatraman, 2023-03-19:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The while loop completes the minute the .q file gets created and does not wait until the q.completed file. This causes the analyze method to not wait. 

I have commented the original line and replaced it with a working line. 

  ```
 if machine == "localhost":
            #while not os.path.exists(queue_file) and not os.path.exists(queue_file_completed):
            while not all([os.path.isfile(f)for f in [queue_file,queue_file_completed]]):
                time.sleep(0.5)
            with open(queue_file, "r") as f:
                lines = f.readlines()
                for line in lines:
                    if "JobID" in line:
                        ls = line.split("=")[1].strip().strip("'")
                        self.last_run_job = ls
                        self.last_run_log = os.path.join(filename + ".batchinfo", project_name + "-" + ls + ".log")
        return True
```

### Steps To Reproduce

open a aedb file and run a sim and make solve_in_batch True

`h3d.analyze("Pyaedt_setup",num_tasks=1,num_cores=8,solve_in_batch=True)`

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.10

### Installed packages

aiohttp==3.8.1
aiosignal==1.2.0
anyio==3.6.1
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
asttokens==2.0.5
async-timeout==4.0.2
attrs==21.4.0
Babel==2.10.3
backcall==0.2.0
beautifulsoup4==4.11.1
bleach==5.0.1
certifi==2022.6.15
cffi==1.15.0
charset-normalizer==2.1.0
clr-loader==0.2.4
contourpy==1.0.7
cycler==0.11.0
debugpy==1.6.2
decorator==5.1.1
defusedxml==0.7.1
distro==1.7.0
dotnetcore2==3.1.23
entrypoints==0.4
executing==0.8.3
fastjsonschema==2.15.3
fonttools==4.34.4
frozenlist==1.3.0
idna==3.3
imageio==2.19.3
ipykernel==6.15.1
ipython==8.4.0
ipython-genutils==0.2.0
ipywidgets==8.0.4
jedi==0.18.1
Jinja2==3.1.2
json5==0.9.8
jsonschema==4.7.2
jupyter==1.0.0
jupyter-client==7.3.4
jupyter-console==6.5.1
jupyter-server==1.18.1
jupyter_core==5.2.0
jupyterlab==3.4.3
jupyterlab-pygments==0.2.2
jupyterlab-server==2.15.0
jupyterlab-widgets==3.0.5
kiwisolver==1.4.3
MarkupSafe==2.1.1
matplotlib==3.7.1
matplotlib-inline==0.1.3
mistune==0.8.4
multidict==6.0.2
nbclassic==0.5.1
nbclient==0.6.6
nbconvert==6.5.0
nbformat==5.4.0
nest-asyncio==1.5.5
notebook==6.5.2
notebook-shim==0.1.0
numpy==1.23.1
packaging==21.3
pandas==1.5.1
pandocfilters==1.5.0
parso==0.8.3
pexpect==4.8.0
pickleshare==0.7.5
Pillow==9.2.0
platformdirs==3.0.0
plumbum==1.7.2
prometheus-client==0.14.1
prompt-toolkit==3.0.30
psutil==5.9.1
ptyprocess==0.7.0
pure-eval==0.2.2
pyaedt==0.6.55
pycparser==2.21
Pygments==2.12.0
pyparsing==3.0.9
pyrsistent==0.18.1
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.1
pyvista==0.35.1
pyzmq==23.2.0
qtconsole==5.4.0
QtPy==2.3.0
requests==2.28.1
rpyc==5.3.0
scikit-rf==0.25.0
scipy==1.10.1
scooby==0.5.12
Send2Trash==1.8.0
six==1.16.0
sniffio==1.2.0
soupsieve==2.3.2.post1
stack-data==0.3.0
style==1.1.0
terminado==0.15.0
tinycss2==1.1.1
tornado==6.2
traitlets==5.9.0
update==0.0.1
urllib3==1.26.10
vtk==9.2.0rc1
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.3.3
widgetsnbextension==4.0.5
wslink==1.6.6
yarl==1.7.2

vishak-venkatraman, 2023-03-19:
Having both means that the sim never returns 
so changing to this 

`while not os.path.exists(queue_file_completed):`

vishak-venkatraman, 2023-03-19:
Seems a bit more involved looking at the function. 
wait until q file created to read it and get job id
then wait until q.completed file created to return to calling function

MaxJPRey, 2023-04-04:
Hi @vishak-venkatraman ,
The PR #2749 fixing this issue has been integrated into the release [0.6.61](https://github.com/pyansys/pyaedt/releases/tag/v0.6.61).

Don't hesitate to re-open the issue if you still experiment any limitations related to this topic.

Thanks for your feedback.

vishak-venkatraman, 2023-04-05:
@maxcapodi78 @MaxJPRey 

It looks like the function wasn't written exactly like I had tested and filed above in the bug (: 
I gave a working code example above after testing. 

```
Job <362024> is submitted to queue <lsfqueue>.
pyaedt ERROR: Error in :   File "/pyaedt_error/venv/lib/python3.10/site-packages/pyaedt/application/Analysis.py", line 1964, in solve_in_batch
pyaedt ERROR: [Errno 2] No such file or directory: '/pyaedt_error/pcb_ftb.aedt.q' - solve_in_batch -  IO ERROR.
pyaedt ERROR: Arguments with values: 
pyaedt ERROR:     filename = None 
pyaedt ERROR:     machine = localhost 
pyaedt ERROR:     run_in_thread = False 
pyaedt ERROR:     num_cores = 48 
pyaedt ERROR:     num_tasks = -1 
pyaedt ERROR:     setup_name = None 
pyaedt ERROR:     revert_to_initial_mesh = False 
pyaedt ERROR: Check Online documentation on: https://aedt.docs.pyansys.com/search.html?q=solve_in_batch
pyaedt ERROR: Error in :   File "/pyaedt_error/venv/lib/python3.10/site-packages/pyaedt/application/Analysis.py", line 1178, in get_sweeps
pyaedt ERROR: 'NoneType' object has no attribute 'GetSweeps' - get_sweeps -  ATTRIBUTE ERROR.
pyaedt ERROR: Arguments with values: 
pyaedt ERROR:     name = Pyaedt_setup 
pyaedt ERROR: Check Online documentation on: https://aedt.docs.pyansys.com/search.html?q=get_sweeps
False
Traceback (most recent call last):
  File "/pyaedt_error/pdn_final.py", line 96, in <module>
    print(h3d.get_sweeps("Pyaedt_setup"))
  File "/pyaedt_error/venv/lib/python3.10/site-packages/pyaedt/application/Analysis3DLayout.py", line 389, in port_list
    return self.oexcitation.GetAllPortsList()
AttributeError: 'NoneType' object has no attribute 'GetAllPortsList'

```
Please cmpare the code I provided and the one here below which is in 0.6.61

.q file will not exist when .q.completed exists and that's why this code errors. 

```
        if machine == "localhost":
            while not os.path.exists(queue_file_completed):
                time.sleep(0.5)
==>         with open(queue_file, "r") as f:
                lines = f.readlines()
                for line in lines:
                    if "JobID" in line:
                        ls = line.split("=")[1].strip().strip("'")
                        self.last_run_job = ls
                        self.last_run_log = os.path.join(filename + ".batchinfo", project_name + "-" + ls + ".log")
```

and please have some basic unit tests run on PRs and release versions

vishak-venkatraman, 2023-04-05:
This is how I fixed it now and tested and it works. 

```
        if machine == "localhost":
            while not os.path.exists(queue_file):
                time.sleep(0.5)
            with open(queue_file, "r") as f:
                lines = f.readlines()
                for line in lines:
                    if "JobID" in line:
                        ls = line.split("=")[1].strip().strip("'")
                        self.last_run_job = ls
                        self.last_run_log = os.path.join(filename + ".batchinfo", project_name + "-" + ls + ".log")
            while not os.path.exists(queue_file_completed):
                time.sleep(0.5)
        return True
```

PR #2749 (2023-03-31): polar_plot_3d_pyvista return Plotter object and fix submit job bug
- If show is False, return Plotter object not shown

- Fix submit job issue

- Add 'all' to toml

Close #2598 
Close #2644