ansys/pyaedt #2648: Add ...

vishak-venkatraman, 2023-03-19:
### Description of the feature

with submit_job and custom areg file

```
Ansys Electronics Desktop Version 2022.2.2, Build: 2022-07-10 21:46:50
Location: ansysEM-22.2/v222/Linux64/ansysedt.exe
Batch Solve/Save: /path/file.aedt
Starting Batch Run: 8:22:43 PM  Mar 17, 2023
Temp directory: /tmp
Project directory: /home/user/Ansoft
[info] Running "LSF Scheduler" job with ID "342219".  Command line: "ansysEM-22.2/v222/Linux64/ansysedt.exe -distributed -machinelist numcores=48 -auto -monitor -useelectronicsppe=1 -ng -batchoptions  'HFSS 3D Layout Design/RAMLimitPercent'=90 -batchsolve /path/file.aedt".
Simulation settings:
^M
Design type: HFSS 3D Layout Design
Allow off core: True
Using automatic settings
Optimetrics variations will be solved sequentially.^M

Machines:
hostname [784998 MB]: RAM: 90%, 48 cores, 0 GPUs^M
```
The interpolating frequency sweep is solved in parallel

```
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency Sweep (9:41:40 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Solution - Sweep1 : Interpolating HFSS Frequency Sweep, Solving Distributed - up to 13 frequencies in parallel, Time: 03/17/2023 21:41:40 (9:41:40 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        From 0Hz to 1GHz, 792 Frequencies (9:41:40 PM  Mar 17, 2023)
[info] Project:project, Design:project, An interpolating frequency sweep with 792 points has been started using HFSS - Solving Distributed. (9:41:40 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Using automatic algorithm to locate minimum frequency for the sweep. (9:41:40 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Solving with MPI (Intel MPI) (9:41:46 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency - 1MHz : hostname (9:41:58 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Distributed Solve Group #1; Automatic determination of minimum frequency (9:41:58 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency - 550kHz : hostname (9:42:00 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Distributed Solve Group #1; Automatic determination of minimum frequency (9:42:00 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency - 100kHz : hostname (9:42:00 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Distributed Solve Group #1; Automatic determination of minimum frequency (9:42:00 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency - 100MHz : hostname (9:42:00 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Distributed Solve Group #1; Automatic determination of minimum frequency (9:42:00 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency - 55MHz : hostname (9:42:00 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Distributed Solve Group #1; Automatic determination of minimum frequency (9:42:00 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency - 10MHz : hostname (9:42:01 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Distributed Solve Group #1; Automatic determination of minimum frequency (9:42:01 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency - 5.5MHz : hostname (9:42:01 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Distributed Solve Group #1; Automatic determination of minimum frequency (9:42:01 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency - 550MHz : hostname (9:42:01 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Distributed Solve Group #1; Automatic determination of minimum frequency (9:42:01 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency - 325MHz : hostname (9:42:01 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Distributed Solve Group #1; Automatic determination of minimum frequency (9:42:01 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency - 775MHz : hostname (9:42:01 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Distributed Solve Group #1; Automatic determination of minimum frequency (9:42:01 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Frequency - 212.5MHz : hostname (9:42:02 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE]        Distributed Solve Group #1; Automatic determination of minimum frequency (9:42:02 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Simulation Setup : Real Time 00:02:23 : CPU Time 00:02:23 : Memory 7.98 G : Disk: 0 Bytes, Tetrahedra: 3196684 (9:57:20 PM  Mar 17, 2023)
[info] Project:project, Pyaedt_Setup : [PROFILE] Matrix Assembly : Real Time 00:04:17 : CPU Time 00:07:29 : Memory 19.9 G : Disk: 0 Bytes, Tetrahedra: 3196684, Lumped ports: 42 (9:57:20 PM  Mar 17, 2023)
```

With solve_in_batch function using analyze method
```
h3d = pyaedt.Hfss3dLayout(specified_version="2022.2",projectname=target_aedt, non_graphical=non_graphical)
h3d.analyze("Pyaedt_setup",num_tasks=1,num_cores=48,solve_in_batch=True)
```

```
Ansys Electronics Desktop Version 2022.2.2, Build: 2022-07-10 21:46:50
Location: ansysEM-22.2/AnsysEM22.2/Linux64/ansysedt.exe
Batch Solve/Save: /path/file.aedt
Starting Batch Run: 10:07:18 PM  Mar 18, 2023
Num tasks for machine localhost = 1
Num cores for machine localhost = 48
RAM limit for machine localhost = 90
GPU count for machine localhost = 1
Temp directory: /tmp
Project directory: /home/user/Ansoft
[info] Running "LSF Scheduler" job with ID "342593".  Command line: "ansysEM-22.2/AnsysEM22.2/Linux64/ansysedt.exe -ng -BatchSolve -machinelist list=localhost:1:48:90%:1 -Monitor /path/file.aedt".
Simulation settings:
^M
Design type: HFSS 3D Layout Design
Allow off core: True
Using manual settings
Two level: Disabled
Distribution types: Variations, Frequencies, Mesh Assembly
Machines:
localhost RAM: 90%, task0:48 cores, num gpus:1, gpu indexes:0
```
Where as here interpolating frequency sweep is not parallel 

```
[info] Project:project, Pyaedt_setup : [PROFILE] Frequency Sweep (11:27:10 PM  Mar 18, 2023)
[info] Project:project, Pyaedt_setup : [PROFILE] Solution - Sweep1 : Interpolating HFSS Frequency Sweep, Time: 03/18/2023 23:27:10 (11:27:10 PM  Mar 18, 2023)
[info] Project:project, Pyaedt_setup : [PROFILE]        From 0Hz to 1GHz, 792 Frequencies (11:27:10 PM  Mar 18, 2023)
[info] Project:project, Design:project, An interpolating frequency sweep with 792 points has been started using HFSS. (11:27:10 PM  Mar 18, 2023)
[info] Project:project, Pyaedt_setup : [PROFILE]        Using automatic algorithm to locate minimum frequency for the sweep. (11:27:10 PM  Mar 18, 2023)
[info] Project:project, Pyaedt_setup : [PROFILE] Frequency - 100MHz : hostname (11:27:15 PM  Mar 18, 2023)
[info] Project:project, Pyaedt_setup : [PROFILE]        HFSS: Single Frequency Solve Group #1; Automatic determination of minimum frequency (11:27:15 PM  Mar 18, 2023
[info] Project:project, Pyaedt_setup : [PROFILE] Simulation Setup : Real Time 00:02:21 : CPU Time 00:02:21 : Memory 7.97 G : Disk: 0 Bytes, Tetrahedra: 3196684 (11:29:39 PM  Mar 18, 2023)
[info] Project:project, Pyaedt_setup : [PROFILE] Matrix Assembly : Real Time 00:04:34 : CPU Time 00:26:47 : Memory 19.8 G : Disk: 0 Bytes, Tetrahedra: 3196684, Lumped ports: 42 (11:34:13 PM  Mar 18, 2023)
```

### Steps for implementing the feature

in the solve_in_batch function add **-distributed** to the options list and if **num_tasks is -1** then append **-auto** to the options. No other changes. 

```
1876         if os.path.exists(queue_file_completed):
1877             os.unlink(queue_file_completed)
1878 
1879         options = [
1880             "-ng",
1881             "-BatchSolve",
1882             "-distributed",
1883             "-machinelist",
1884             "list={}:{}:{}:90%:1".format(machine, num_tasks, num_cores),
1885             "-Monitor",
1886         ]
1887         if num_tasks == -1:
1888             options.append("-auto")
1889         if setup_name and design_name:
1890             options.append(
1891                 "{}:{}:{}".format(
1892                     design_name, "Nominal" if setup_name in self.setup_names else "Optimetrics", setup_name
1893                 )
1894             )
```

Then user can call with a -1 for num_tasks to enable distributed frequency sweep. if they don't call out num_tasks it is set as 1 so it should revert to normal as it is today. 

```
h3d = pyaedt.Hfss3dLayout(specified_version="2022.2",projectname=target_aedt, non_graphical=non_graphical)
h3d.analyze("Pyaedt_setup",num_tasks=**-1**,num_cores=48,solve_in_batch=True)
```

This worked Feel free to make it a better way

MaxJPRey, 2023-03-27:
@maxcapodi78 I think we can close this issue now that #2653 has been merged. What do you think?

closed 2023-03-28 by Samuelopez-ansys

vishak-venkatraman, 2023-04-03:
@maxcapodi78 please tag with release versions so that I can follow up and verify

MaxJPRey, 2023-04-04:
@vishak-venkatraman  PR #2653 was integrated in the release [v0.6.57](https://github.com/pyansys/pyaedt/releases/tag/v0.6.57). If you try with a newer release, it should be implemented too.