ansys/pyaedt #2649: Can't work with AEDT versions 2022 R2 but OK with AEDT versions 2021 R2.

wahahahachin, 2023-03-20:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

A tool that developed by pyaedt for Hfss3dLayout application. It can be executed in AEDT versions 2021 R2. But, with the same code, it can not be excecuted successfully in 2022 R2 (can open Hfss3dLayout, but can not create stackup, via, line ...etc). 

### Steps To Reproduce

Use pyaedt.Hfss3dLayout, It will success in 2021 R2. But fail in 2022 R2 (only open Hfss3dLayout, but can't run the code to create stackup, via...etc).

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

altgraph                  0.17.3
cffi                      1.15.1
clr-loader                0.2.5
et-xmlfile                1.1.0
numpy                     1.24.2
openpyxl                  3.1.2
pefile                    2023.2.7
pip                       23.0.1
plumbum                   1.8.1
psutil                    5.9.4
pyaedt                    0.6.56
pycparser                 2.21
pyinstaller               5.9.0
pyinstaller-hooks-contrib 2023.1
pythonnet                 3.0.1
pywin32                   305
pywin32-ctypes            0.2.0
rpyc                      5.3.0
setuptools                57.4.0

maxcapodi78, 2023-03-22:
@wahahahachin can you provide more details or the code that is failing so we can try to reproduce it?

wahahahachin, 2023-03-22:
Dear Massimo Capodiferro,

Please find a simple code in the attached, which is to call HFSS3DL
and create a stackup.
It works in the PC that is with AEDT_2021R2, but fails in that PC that is
with AEDT_2022R2 (can successfully open HFSS3DL only, but can't run the
code to create stackup).

BRs,
Yuchin

Samuelopez-ansys, 2023-03-31:
Hi @wahahahachin ,

I can not see the code. Please could you share it again in GitHub directly?

Thank you

wahahahachin, 2023-04-13:
Dear Samuel Lopez,
Please find a simple code in the attached, which is to call HFSS3DL and create a stackup.It works in the PC that is with AEDT_2021R2, but fails in that PC that is with AEDT_2022R2 (can successfully open HFSS3DL only, but can't run the code to create stackup). 
[Exmaple_Stackup_in_3DLayout (1).zip](https://github.com/pyansys/pyaedt/files/11217907/Exmaple_Stackup_in_3DLayout.1.zip)

Samuelopez-ansys, 2023-04-13:
Hi @wahahahachin ,

It looks like this AEDT API method does not work in 2022R2, it is not a PyAEDT problem, it is an AEDT issue.

BUT, I think you should use EDB to import/load ECAD. For example for your case, to load an XML stackup, it is very simple:

```
from pyaedt import Edb
aedb_project = Edb("test.aedb", edbversion="2022.2")
aedb_project.core_stackup.load("stackup.xml")
aedb_project.save_edb()
aedb_project.close_edb()

h3d = Hfss3dLayout(new_desktop_session=True, non_graphical=False,
                   projectname="test.aedb", close_on_exit=True,
                   specified_version="2022.2")
h3d.save_project()
h3d.release_desktop(False, False)
```

Please check the examples, where you will find the power of EDB API: 
https://aedt.docs.pyansys.com/version/stable/examples/00-EDB/index.html

If it solves your questions, please close this issue.

gmalinve, 2023-06-21:
hi @wahahahachin ,

does the code snippet provided by @Samuelopez-ansys solve your issue?
Please do let me know and in case close the issue.
Thanks so much!

Giulia

wahahahachin, 2023-06-26:
Hi @gmalinve,

Thanks! But, no. I had tried it but still can't import the stackup in XML format in this simple case.

Yuchin.

Samuelopez-ansys, 2023-07-17:
@wahahahachin I tried with the zip you shared and the code I shared and it works, please check you have the latest PyAEDT version, and I encourage you to install the lastest AEDT version too.

closed 2023-09-04 by Samuelopez-ansys