ansys/pyaedt #2654: Add Temperature Dependent Variation for Thermal Blocks in Icepak

StiftMC, 2023-03-20:
### Description of the feature

Temperature Dependent Variation of Thermal Blocks is already in AEDT GUI, but can't be accessed via pyAEDT's create_source_block function. 

[image]

### Steps for implementing the feature

Add thermal_dependent_dataset parameter to function create_source_block, similar to create_source_power function.

Samuelopez-ansys, 2023-05-11:
@lorenzovecchietti I think you have already implement it, right?

lorenzovecchietti, 2023-05-17:
> @lorenzovecchietti I think you have already implemented it, right?

It is implemented just in `create_source_power`. Unfortunately this is a common issue of most Icepak bc assignment function and I'm currently trying so update/replace them all.

`create_source_block` will be the next :)

PR #3018 (2023-05-19): New block assignment function
Close #2654
closed 2023-05-19 by maxcapodi78