ansys/pyaedt #2656: pip install pyaedt should install dependent modules

vishak-venkatraman, 2023-03-20:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

if I pip install pyaedt and run sim in command line mode without Jupyter it warns of these missing modules. I feel that they should be installed as part of pyaedt and one should not assume Jupyterlab or Jupyter notebook would be installed already 

/test2/pyaedt_env/lib/python3.10/site-packages/pyaedt/edb_core/edb_data/components_data.py:12: UserWarning: The NumPy module is required to run some functionalities of EDB.
Install with 

pip install numpy

Requires CPython.
  warnings.warn(
/test2/pyaedt_env/lib/python3.10/site-packages/pyaedt/generic/plot.py:21: UserWarning: The NumPy module is required to run some functionalities of PostProcess.
Install with 

pip install numpy

Requires CPython.
  warnings.warn(
/test2/pyaedt_env/lib/python3.10/site-packages/pyaedt/generic/plot.py:31: UserWarning: The PyVista module is required to run some functionalities of PostProcess.
Install with 

pip install pyvista

Requires CPython.
  warnings.warn(
/test2/pyaedt_env/lib/python3.10/site-packages/pyaedt/generic/plot.py:42: UserWarning: The Matplotlib module is required to run some functionalities of PostProcess.
Install with 

pip install matplotlib

Requires CPython.
  warnings.warn(

### Steps To Reproduce

create a new virtual env and just do pip install pyaedt and run a sim. 

```
python3.10 -m venv venv
source vev/bin/activate
pip install --upgrade pip
pip install wheel
pip install pyaedt

```

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.10

### Installed packages

cffi==1.15.0
clr-loader==0.2.5
distro==1.8.0
dotnetcore2==3.1.23
plumbum==1.8.1
psutil==5.9.4
pyaedt==0.6.56
pycparser==2.21
pythonnet==3.0.1
rpyc==5.3.0

vishak-venkatraman, 2023-03-22:
@Samuelopez-ansys  that's ok but there should be a way to suppress the warning as I don't want it in my log file each time. Is there a way to do it.

Samuelopez-ansys, 2023-03-23:
@vishak-venkatraman  Which warnings do you have?

I am testing the new pyaedt version and I don't see any warning:

```
import pyaedt

pyaedt.settings.use_grpc_api = False

app = pyaedt.Hfss(specified_version="2022.2", non_graphical=False, new_desktop_session=True)

app.release_desktop(False, False)
```

vishak-venkatraman, 2023-03-24:
@Samuelopez-ansys I get the messages I show above when I filed the bug. Take a look at the Bug description. Also Try running this example https://aedt.docs.pyansys.com/release/0.6/examples/00-EDB/09_Configuration.html

Samuelopez-ansys, 2023-03-24:
Hi @vishak-venkatraman ,

In this example is normal to see warnings because you are using methods that are using 'advanced'  functionalities which need numpy, matplolib... for example components_data.py, is at the beginning of import numpy and if it does not exist it returns a warning, this is expected behavior.

You could improve the documentation of the example mentioning it is needed full pyaedt documentation for running this example.

Again you just need to do pip install pyaedt .[full] 
And it will install all dependencies.

Please close the issue if you think it could be closed.

Samuelopez-ansys, 2023-03-24:
Sorry the command to install all dependencies is:

pip install pyaedt[full]

cutelolly, 2023-03-24:
It has been updated to the corresponding readme document, please refer to https://github.com/cutelolly/pyaedt/blob/main/README_CN.md

获取Outlook for Android<https://aka.ms/AAb9ysg>

closed 2023-03-27 by vishak-venkatraman