ansys/pyaedt #2675: Bug located in sym-link creation of pyaedt\pyaedt\misc\aedtlib_personalib_install.py

kbhagat-ansys, 2023-03-21:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The `aedtlib_personalib_install.py` is flawed when creating a sym-link to pyaedt on Linux. 
Currently, it checks if pyaedt symlink is present and it won't create it. Since the python virtual environments are version specific, it's best if we delete the sym-links and recreate it every time we run this script. Customers will end up running Install PyaEDT for every new version they install - thus we want this to reflect. 

### Steps To Reproduce

run aedtlib_personalib_install.py on Linux through the Install PyAEDT script from AEDT

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

N/A

Samuelopez-ansys, 2023-11-07:
@kbhagat-ansys Could you help us and start a pull request with this modification? We will help you if you need further assistance.

kbhagat-ansys, 2023-11-08:
> @kbhagat-ansys Could you help us and start a pull request with this modification? We will help you if you need further assistance.

@Samuelopez-ansys  Sounds good. Will look into it later this week or next.

closed 2024-05-14 by Samuelopez-ansys