ansys/pyaedt #2676: Bug located in ...

stephennewberry, 2023-03-21:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Any use of the `export_model_obj` function with an object list or file path used as input causes PyAEDT to crash (the pop-up with "do you want to save your data"). Passing only the `export_as_single_objects` arg does not cause this behavior.

### Steps To Reproduce

Call `export_model_obj` on an HFSS (classic) design while passing either a file path or an object list (or both).

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

anyio==3.6.2
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
arrow==1.2.3
asttokens==2.2.1
attrs==22.2.0
backcall==0.2.0
beautifulsoup4==4.11.2
bleach==6.0.0
cadquery==2.2.0
cadquery-ocp==7.7.0
casadi==3.5.5
cffi==1.15.1
clr-loader==0.2.5
colorama==0.4.6
comm==0.1.2
debugpy==1.6.6
decorator==5.1.1
defusedxml==0.7.1
executing==1.2.0
ezdxf==1.0.2
fastjsonschema==2.16.3
fqdn==1.5.1
idna==3.4
importlib-metadata==6.1.0
ipykernel==6.21.3
ipython==8.11.0
ipython-genutils==0.2.0
ipywidgets==8.0.4
isoduration==20.11.0
jedi==0.18.2
Jinja2==3.1.2
jsonpointer==2.3
jsonschema==4.17.3
jupyter==1.0.0
jupyter-console==6.6.3
jupyter-events==0.6.3
jupyter_client==8.0.3
jupyter_core==5.3.0
jupyter_server==2.5.0
jupyter_server_terminals==0.4.4
jupyterlab-pygments==0.2.2
jupyterlab-widgets==3.0.5
MarkupSafe==2.1.2
matplotlib-inline==0.1.6
mistune==2.0.5
multimethod==1.9.1
nbclassic==0.5.3
nbclient==0.7.2
nbconvert==7.2.10
nbformat==5.7.3
nest-asyncio==1.5.6
nlopt==2.7.1
notebook==6.5.3
notebook_shim==0.2.2
nptyping==2.0.1
numpy==1.24.2
packaging==23.0
pandas==1.5.3
pandocfilters==1.5.0
parso==0.8.3
path==16.6.0
pickleshare==0.7.5
platformdirs==3.1.1
plumbum==1.8.1
prometheus-client==0.16.0
prompt-toolkit==3.0.38
psutil==5.9.4
pure-eval==0.2.2
pyaedt==0.6.56
pycparser==2.21
Pygments==2.14.0
pyparsing==3.0.9
pyrsistent==0.19.3
python-dateutil==2.8.2
python-json-logger==2.0.7
pythonnet==3.0.1
pytz==2022.7.1
pywin32==305
pywinpty==2.0.10
PyYAML==6.0
pyzmq==25.0.1
qtconsole==5.4.1
QtPy==2.3.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpyc==5.3.0
Send2Trash==1.8.0
six==1.16.0
sniffio==1.3.0
soupsieve==2.4
stack-data==0.6.2
terminado==0.17.1
tinycss2==1.2.1
tornado==6.2
traitlets==5.9.0
typing_extensions==4.5.0
typish==1.9.3
uri-template==1.2.0
wcwidth==0.2.6
webcolors==1.12
webencodings==0.5.1
websocket-client==1.5.1
widgetsnbextension==4.0.5
zipp==3.15.0

Samuelopez-ansys, 2023-03-22:
Hi @stephennewberry,

Please could you attach the snippet code? I can not reproduce the issue:

I have created a simple HFSS design with a box called "Box1", I have launched this script:

```
import pyaedt

pyaedt.settings.use_grpc_api = False

app = pyaedt.Hfss(specified_version="2023.1", non_graphical=False, new_desktop_session=False)

x = app.post.export_model_obj(["Box1"])

app.release_desktop(False, False)
```

And it worked.

stephennewberry, 2023-04-07:
Hi @Samuelopez-ansys, my apologies for my delayed response. I went to reproduce the bug and am no longer able to see this issue. I assume this was an error in my configuration then. Sorry again - and thank you for the quick response!

closed 2023-04-07 by stephennewberry