ansys/pyaedt #2704: How to I get the larger of two faces or both faces with top_face_z

jpotter0, 2023-03-25:
I have a model of a resonant cavity that has two faces on each end of its z axis. Of the two faces one is small and has low field so its boundary condition doesn't matter. The larger faces need to be set to symmetrical H. I am iterating through 15 resonators, each slightly longer than the previous. For the first 8 iterations top_face_z returns the larger of the faces. Then on the 9th iteration it returns the smaller face. The problem is that I am subtracting these faces from the list of faces so I can make the boundary conductive wherever the faces are not symmetry H. The models are all similar so I could manually identify the faces I wish to make Symmetry H . However, I discovered that if the faces were 254 for the larger face and 255 for the smaller face, for example, there would be an iteration where the numbers are reversed.

What happens with either method is that I end up on some cavities with a conducting boundary where i should have a  Symmetry H boundary.

Ideally, I'd like to get the ID for both snd know which is the larger so I can assign the appropriate boundary conditions. In this case it would be enough to get both ID's and set them both to the same boundary condition. I don't see how to do that.

Here is the statement to find the end boundary:

    top_face_z = aedtapp.modeler.primitives[obj_name].top_face_z

This gets me only one face, but it's not always the face that I want.

How do I get both faces?

fwlfHS, 2023-03-28:
A few methods you could look at:

- first get all faces of the object: [faces](https://aedt.docs.pyansys.com/release/0.6/API/_autosummary/pyaedt.modeler.cad.object3d.Object3d.faces.html#pyaedt.modeler.cad.object3d.Object3d.faces)
- check which faces are on the bounding box: [is_on_bounding](https://aedt.docs.pyansys.com/release/0.6/API/_autosummary/pyaedt.modeler.cad.elements3d.FacePrimitive.is_on_bounding.html#pyaedt.modeler.cad.elements3d.FacePrimitive.is_on_bounding)
- compare the sizes of the faces using [area](https://aedt.docs.pyansys.com/release/0.6/API/_autosummary/pyaedt.modeler.cad.elements3d.FacePrimitive.area.html#pyaedt.modeler.cad.elements3d.FacePrimitive.area)

In addition there are methods like [faces_by_area](https://aedt.docs.pyansys.com/release/0.6/API/_autosummary/pyaedt.modeler.cad.object3d.Object3d.faces_by_area.html#pyaedt.modeler.cad.object3d.Object3d.faces_by_area) if you know the size of the face you are looking for, or [largest_face](https://aedt.docs.pyansys.com/release/0.6/API/_autosummary/pyaedt.modeler.cad.object3d.Object3d.largest_face.html#pyaedt.modeler.cad.object3d.Object3d.largest_face) if the largest face of the object is at the boundary condition.

jpotter0, 2023-03-30:
Thanks for the response! I've managed a workaround, specific to my case, that will do for now.

closed 2023-03-30 by jpotter0