ansys/pyaedt #2717: Improve cutout such for all Component associated to a model keep the same number of pin as defined in the model

amichel0205, 2023-03-27:
### Description of the feature

All component associated to a model(spice):

- Identified component pin associated to the model
- preserve these pins from the cutout

svandenb-dev, 2023-04-18:
@amichel0205 can we close this issue ? was it addressed with the new smart cutout ?

amichel0205, 2023-04-18:
@svandenb-dev @gmalinve 

Giulia ask me the same question this morning. I did not have time yet to check it with a real example.
I supposed from discussion with Massimo that the new cutout method should address it.
So would it be possible to give me 1 or 2 weeks more such I can test it and close it if working fine?

svandenb-dev, 2023-05-30:
closing this issue. New methods has been deployed since several weeks and seems to be working. Please submit an new issue if any error happen.

closed 2023-05-30 by svandenb-dev