ansys/pyaedt #2761: Antennas from json file can only be rotated around one axis if same rotation degree

sara-tell, 2023-04-03:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When using hffs.create_sbr_radar_from_json the antennas are only rotated around the first given axis in the json file if the rotation angle is the same. The issue is in__init__ in parts.py. rotations.index(a) will always just return the first index if the values are the same.

            for a in rotations:
                if rotations_axis[rotations.index(a)].lower() == "x":  # roll
                    r = a
                    self.rot_axis[2] = True
                elif rotations_axis[rotations.index(a)].lower() == "y":  # pitch
                    p = a
                    self.rot_axis[1] = True
                elif rotations_axis[rotations.index(a)].lower() == "z":  # yaw
                    y = a
                    self.rot_axis[0] = True

### Steps To Reproduce

pyaedt.hfss.create_sbr_radar_from_json(radar_file="path_to_ffd_files",
            radar_name="radar.json",
            offset=[0, 0, 0])

Example json file:

{
	"name": "Radar",
	"version": 1,
	"number_tx":"1",
	"number_rx":"0",
	"units":"mm",
	"antennas": {
		"tx1": {
			"antenna_type":"ffd",
			"ffd_name":"Tx1",
			"mode":"tx",
			"offset":["0.0" ,"0.0" ,"0.0"],
			"rotation_axis":"X,Y",
			"rotation":"90deg,90deg"
			}
         }
}

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

cffi==1.15.1
clr-loader==0.2.5
numpy==1.24.2
open-simulation-interface @ file:///C:/Git/open-simulation-interface
plumbum==1.8.1
protobuf==4.22.1
psutil==5.9.4
pyaedt==0.6.57
pycparser==2.21
pythonnet==3.0.1
pywin32==306
rpyc==5.3.0

maxcapodi78, 2023-04-04:
@sara-tell  the PR #2773 should fix your issue. Let me know if you still have the problem

sara-tell, 2023-04-04:
I cannot get my pip to update pyaedt when the version is the same. I believe it should work though. But a more pythonic way of writing it would be

            for i, a in enumerate(rotations):
                if rotations_axis[i].lower() == "x":  # roll
                    r = a
                    self.rot_axis[2] = True
                elif rotations_axis[i].lower() == "y":  # pitch
                    p = a
                    self.rot_axis[1] = True
                elif rotations_axis[i].lower() == "z":  # yaw
                    y = a
                    self.rot_axis[0] = True

closed 2023-04-06 by maxcapodi78