ansys/pyaedt #2768: Improve Edb.modeler.primitives

dcrawforAtAnsys, 2023-04-04:
### Description of the current limitations

``primitives`` is a collection.  Each time the prop is accessed, the full list is recreated. Each item in the list is an instance of ``Edb``.  It would be far more efficient to create an internal ``self._prims`` property and update that every time a primitive is created or deleted.  The difficult part will be to catch when primitives are deleted.

### Code sample expected

create ``self._prims`` as a list or instance of a collection class.  The latter would be more efficient.
Update items as primitives are created or deleted.
Note that each item in ``Edb.modeler.primitives`` is an instance of ``edb_core.edb_data.primitives_data.EdbPrimitives``. Here the naming is confusing since the latter is not a collection.  It should probably be renamed to singular.

### Useful links and references

https://docs.python.org/3/library/collections.html

dcrawforAtAnsys, 2023-04-04:
I think the current class implementation ``EdbPrimitives`` should be renamed to ``EdbPrimitive`` and a new collection class called ``EdbPrimitives`` should be created as a container for instances of the former.

maxcapodi78, 2023-04-06:
Edb is very efficient in collection handling. It can handle very efficiently for us.

closed 2023-04-06 by maxcapodi78