ansys/pyaedt #2779: disable_rlc_component is not working

vishak-venkatraman, 2023-04-04:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am trying to disable a RLC component. The method returns true but the component still is enabled. 

`r = edbapp.core_components.disable_rlc_component("C3")`

My goal is to disable any component Resistor, Inductor, capacitor, IC, IO, Other. But for now I am using this as this is the only API I could find.  

[image]

[image]

[image]

### Steps To Reproduce

open EDB 

disable a component

Save EDB and open in ansysedt

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.10

### Installed packages

NA

vishak-venkatraman, 2023-04-04:
BTW This works but please fix the EDB and also have this feature in EDB

`h3d.modeler.components["C1"].enabled = False`

maxcapodi78, 2023-04-06:
@vishak-venkatraman have you tried:
edbapp.core_components.deactivate_rlc_component(component_name)

the disable_rlc_component disable the R, L or C values

maxcapodi78, 2023-04-06:
@vishak-venkatraman I've added a property is_enabled to the components in Edb with getter and setter

vishak-venkatraman, 2023-04-06:
@maxcapodi78 
1. 
`edbapp.core_components.deactivate_rlc_component(component_name)`

Is not working. I see a warning but when I open the edb I see them as enabled. 

```
pyaedt WARNING: RLC properties for Component C1 has been assigned.
pyaedt WARNING: RLC properties for Component C2 has been assigned.
```
[image]

2. 
I don't see a way in this API to disable only R, or only L or only C from the code 
`r = edbapp.core_components.disable_rlc_component("C3")`

It seems to disable all of them 
[image]

But even that doesn't work.

maxcapodi78, 2023-04-11:
@vishak-venkatraman  you are right. There was a bug that I've fixed in 0.6.70. and I've added also the getters and setters

```
edb.components.instances["C1"].is_enabled = False
```

maxcapodi78, 2023-04-17:
Implemented in v0.6.71

closed 2023-04-17 by maxcapodi78