ansys/pyaedt #2781: Bug located in edb_core.padstack.py

dcrawforAtAnsys, 2023-04-05:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The first argument is the position, which is a list of float or string.  
The position argument from _edb is
```python
position = self._edb.Geometry.PointData(self._get_edb_value(position[0]), self._get_edb_value(position[1]))
```
But position itself is a list, so it should be
```python
  position = self._edb.Geometry.PointData(self._get_edb_value(position[0][0]),
                                          self._get_edb_value(position[0][1]))
```

### Steps To Reproduce

Pass a 2 element list to ``edb.padstack.place()``. The resulting position will be incorrect.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

all pyaedt standard packages.

dcrawforAtAnsys, 2023-04-05:
The issue was related to a subtlety in the way variables are passed to Python in a function. This is not a bug.

maxcapodi78, 2023-04-14:
@dcrawforAtAnsys are you sure? I think it is correct the method.