ansys/pyaedt #2783: export_model_picture function ignores visualization settings

StiftMC, 2023-04-05:
PCBs are shown without copper information in Icepak by default. However, copper visualization can be switched on via right click on 3D component -> Properties... -> Visualization tab or via oEditor scripting.

But even with copper visualization switched on, export_model_picture function delivers only default view of the PCB:

3DModeler
[image]

export_model_picture output:
[image]

Samuelopez-ansys, 2023-05-11:
Hi @StiftMC 

Are you using the latest PyAEDT version?

First of all, you can enable/disable the layers using these commands:

1. You can get the layers (last elements after "Minimum Resolution/Choices"):

`self.aedtapp.odesign.GetChildObject("3D Modeler").GetChildObject(comp.name).GetPropNames()`

Result:
00 = {str} 'Name'
01 = {str} 'Group'
02 = {str} 'Definition Units'
03 = {str} 'Reference Coordinate System'
04 = {str} 'Target Coordinate System'
05 = {str} 'Target Coordinate System/Choices'
06 = {str} 'Type'
07 = {str} 'Minimum Resolution'
08 = {str} 'Minimum Resolution/Choices'
**09 = {str} 'TOP'
10 = {str} 'UNNAMED_002'
11 = {str} 'PWR'
12 = {str} 'UNNAMED_004'
13 = {str} 'LYR_1'
14 = {str} 'UNNAMED_006'
15 = {str} 'LYR_2'
16 = {str} 'UNNAMED_008'
17 = {str} 'GND'
18 = {str} 'UNNAMED_010'
19 = {str} 'BOTTOM'**

2. Enable/Disable each layer:

Disable
self.aedtapp.odesign.GetChildObject("3D Modeler").GetChildObject(comp.name).SetPropValue("TOP", 0)
Enable
self.aedtapp.odesign.GetChildObject("3D Modeler").GetChildObject(comp.name).SetPropValue("TOP", 1)

Then I exported the picture using:

`aedtapp.post.export_model_picture()`

And it exported the same view I had in AEDT:

[image]

Samuelopez-ansys, 2023-05-30:
@StiftMC Could you update Pyaedt version and try again?

Please provide in a zip the script and the dummy project. I am not sure AEDT API allows to control of all UI elements.

StiftMC, 2023-06-06:
Hi @Samuelopez-ansys, 

do you have any updates here?

Samuelopez-ansys, 2023-06-08:
Hi @StiftMC ,

I debug your case, and the problem is not in PyAEDT. It is the AEDT API, when you disable show_region, it is also disabling the PCB layers, so from PyAEDT we can not fix it. I will submit a bug internally in Ansys.

One workaround if it is urgent your problem, is removing the Region, and Undo or create a new region (taking the region size before removing):

ipk.modeler["Region"].delete()

ipk.post.export_model_picture('D:/Temp/test.jpg',
                                  show_axis=False,
                                  show_grid=False,
                                  show_ruler=False,
                                  show_region=True,
                                  selections=pcb,
                                  orientation="isometric",
                                  width=1920,
                                  height=1080)

ipk.odesign.Undo()
ipk.odesign.Undo()
ipk.odesign.Undo()

StiftMC, 2023-06-16:
Hi @Samuelopez-ansys, 

deleting and undo operation worked for me. Thanks! 

I think there is also a bug in selections. If I select any component in selections, it will plot only this component. If I do the same with the PCB 3D component, it plots the PCB and the component mesh region. I'd expect that it only plots the PCB, as it would do with components/primitives. 

However, I overcome this also with the deleting and undo workaround.  

Thanks for your support.

closed 2023-06-16 by Samuelopez-ansys