ansys/pyaedt #2801: Fail to open project with win32com.client

smpourjaafari, 2023-04-07:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I design a waveguide filter in HFSS and I want to generate its digital twin. is it possible to use this method:

from pyaedt import TwinBuilder
app = TwinBuilder("Path To Project")

### Steps To Reproduce

Window Terminal
Command Prompt
jupyter lab

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

aiofiles==22.1.0
aiosqlite==0.18.0
anyio==3.6.2
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
arrow==1.2.3
asttokens==2.2.1
attrs==22.2.0
Babel==2.12.1
backcall==0.2.0
beautifulsoup4==4.12.0
bleach==6.0.0
certifi==2022.12.7
cffi==1.15.1
charset-normalizer==3.1.0
click==8.1.3
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.5
colorama==0.4.6
comm==0.1.3
contourpy==1.0.7
cycler==0.11.0
debugpy==1.6.6
decorator==5.1.1
defusedxml==0.7.1
et-xmlfile==1.1.0
executing==1.2.0
fastjsonschema==2.16.3
Fiona==1.9.2
fonttools==4.39.3
fqdn==1.5.1
geopandas==0.12.2
idna==3.4
imageio==2.27.0
ipykernel==6.22.0
ipython==8.12.0
ipython-genutils==0.2.0
ipywidgets==8.0.6
isoduration==20.11.0
jedi==0.18.2
Jinja2==3.1.2
json5==0.9.11
jsonpointer==2.3
jsonschema==4.17.3
jupyter==1.0.0
jupyter-console==6.6.3
jupyter-events==0.6.3
jupyter-ydoc==0.2.3
jupyter_client==8.1.0
jupyter_core==5.3.0
jupyter_server==2.5.0
jupyter_server_fileid==0.8.0
jupyter_server_terminals==0.4.4
jupyter_server_ydoc==0.8.0
jupyterlab==3.6.3
jupyterlab-pygments==0.2.2
jupyterlab-widgets==3.0.7
jupyterlab_server==2.22.0
kiwisolver==1.4.4
MarkupSafe==2.1.2
matplotlib==3.7.1
matplotlib-inline==0.1.6
mistune==2.0.5
munch==2.5.0
nbclassic==0.5.4
nbclient==0.7.3
nbconvert==7.2.10
nbformat==5.8.0
nest-asyncio==1.5.6
networkx==3.0
notebook==6.5.3
notebook_shim==0.2.2
numpy==1.24.1
openpyxl==3.0.10
osmnx==1.3.0
packaging==23.0
pandas==1.5.3
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.5.0
platformdirs==3.2.0
plumbum==1.8.1
pooch==1.7.0
prometheus-client==0.16.0
prompt-toolkit==3.0.38
psutil==5.9.4
pure-eval==0.2.2
pyaedt==0.6.61
pycparser==2.21
Pygments==2.14.0
pyparsing==3.0.9
pyproj==3.5.0
pyrsistent==0.19.3
python-dateutil==2.8.2
python-json-logger==2.0.7
pythonnet==3.0.1
pytz==2023.3
pyvista==0.37.0
pywin32==306
pywinpty==2.0.10
PyYAML==6.0
pyzmq==25.0.2
qtconsole==5.4.2
QtPy==2.3.1
requests==2.28.2
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpyc==5.3.0
scikit-rf==0.26.0
scipy==1.10.1
scooby==0.7.1
Send2Trash==1.8.0
shapely==2.0.1
six==1.16.0
sniffio==1.3.0
soupsieve==2.4
SRTM.py==0.3.7
stack-data==0.6.2
terminado==0.17.1
tinycss2==1.2.1
tomli==2.0.1
tornado==6.2
traitlets==5.9.0
uri-template==1.2.0
urllib3==1.26.15
utm==0.7.0
vtk==9.2.6
wcwidth==0.2.6
webcolors==1.13
webencodings==0.5.1
websocket-client==1.5.1
widgetsnbextension==4.0.7
y-py==0.5.9
ypy-websocket==0.8.2

SushobhitGupt, 2023-04-13:
Hello,

Yes, you can use the method as described,

Do Check out the syntax for the Command arguments.

tb = pyaedt.TwinBuilder(projectname=pyaedt.generate_unique_project_name(),
                        specified_version=[desktop_version],
                        non_graphical=false,
                        new_desktop_session=true
                        )

The User Guide is a good place to Start:
https://aedt.docs.pyansys.com/version/stable/User_guide/index.html

Here is a Twin Builder Specific Example:
https://aedt.docs.pyansys.com/version/stable/examples/07-TwinBuilder/02-Wiring_A_Rectifier.html#sphx-glr-examples-07-twinbuilder-02-wiring-a-rectifier-py

If you have any specific error please mention it in the comments

Hope that Helps
-Sushobhit

SushobhitGupt, 2023-04-24:
Closing the issue, please reopen in case of additional questions.