ansys/pyaedt #2804: How do I get the imaginary component of frequency? I need to calculate the Q.

jpotter0, 2023-04-10:
Here is the code snippet:

    setup = "Setup"+str(cellnum)
    running = aedtapp.analyze_setup(setup)
    post = aedtapp.post
    data = post.get_solution_data("re(Mode(1))")
    f_Hz = data.data_real()
    df_Hz = data.data_imag()
    fff = float(f_Hz[0])

I have the lossy surfaces defined using the Python script from HFSS. The method of using objects made from surfaces doesn't work for me because the curvilinear coordinated are turned in HFSS for these objects which affect frequency accuracy.

I need to calculate the Q, or get the Q directly.

Jim

Samuelopez-ansys, 2023-04-11:
Hi @jpotter0 ,

The problem of getting 're(Mode(1))' is you are retrieving the real part of Mode(1), so you are not getting the imag. part. 

Try:

data_im = post.get_solution_data("im(Mode(1))")
data_im .data_real()

OR another option could be:

data = post.get_solution_data("Mode(1)")

But I am not sure if this last option works or not, please let me know.

jpotter0, 2023-04-11:
Thanks, Samuel! I'll check it out and get back to you.- Jim

jpotter0, 2023-04-12:
> it on GitHub, or

Samuel,

The attached screen shot shows what works. I 
don't know why there's no data on the f_Hz line, 
but it works in the calculation when I get the frequency in MHz.

Best regards,

Jim

jpotter0, 2023-04-12:
It looks like the screenshot didn't show up in the email answer. Here it is:

[image]

Samuelopez-ansys, 2023-04-14:
Hi @jpotter0 any update?

jpotter0, 2023-04-21:
a modification of your suggestion cleared up the problem.

closed 2023-04-21 by jpotter0