ansys/pyaedt #2827: Bug located in SYZ frequency sweep setup

pyAnsysUser, 2023-04-13:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When adding SYZ frequency sweep, there is a default frequency range that I did not define from 0-20GHz.

Additionally:
- the log sweep creates multiple 2-point sweeps
- the single DC point is not accurately depicted
- The linear step sweep is converted to a linear count sweep

### Steps To Reproduce

```
from pyaedt import Edb
import os.path
working = $user_file_path$ #user path omitted
temp_dir = os.path.join(working, r'Galileo.aedb')
edb = Edb(edbpath=temp_dir, edbversion="2022.2")
new_path = os.path.join(working, 'test' )
edb.save_edb_as(os.path.join(new_path, r'syz.aedb'))

#add port
edb.core_siwave.create_circuit_port_on_net("U1B5","V1P5_S3", port_name = "syzport")

# setup frequency sweep
setup1 = edb.create_siwave_syz_setup("syz_setup")

setup1.add_frequency_sweep(name = "mysweep",frequency_sweep=[
                          ["linear count", "0", "0", 1], 
                          ["log scale", "100Hz", "20MHz", 30], 
                          ["linear scale", "20MHz", "2GHz", "10MHz"], 
                          ])

setup1.use_si_settings = False #use PI radiobutton. Produces warning during solve but desired setting is used
edb.core_siwave.create_exec_file(add_syz = True)
edb.save_edb()
edb.solve_siwave()
edb.close_edb()
```

expect:
linear count 0 Hz, 1 point
log 100Hz - 20MHz, 30 points per decade
linear step 20MHz - 2 GHz, 10MHz steps

Actual:
[image]
[image]

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

certifi==2022.12.7
cffi==1.15.1
charset-normalizer==3.1.0
clr-loader==0.2.5
cycler==0.11.0
fonttools==4.38.0
idna==3.4
imageio==2.27.0
kiwisolver==1.4.4
matplotlib==3.5.3
numpy==1.21.6
packaging==23.1
Pillow==9.5.0
platformdirs==3.2.0
plumbum==1.8.1
pooch==1.7.0
psutil==5.9.4
pyaedt==0.6.70
pycparser==2.21
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.1
pyvista==0.38.5
pywin32==306
requests==2.28.2
rpyc==5.3.0
scooby==0.7.1
six==1.16.0
typing-extensions==4.5.0
urllib3==1.26.15
vtk==9.2.6

maxcapodi78, 2023-04-14:
Hi @pyAnsysUser I've fixed a bug on Freq Sweep yesterday in #2817. We were not clearing the sweep data before adding more points. Give it a try in v0.6.71 which I will create today and reopen the issue

closed 2023-04-14 by maxcapodi78

pyAnsysUser, 2023-04-14:
Yes, this merge has removed the 0-20GHz sweep that was observed previously