ansys/pyaedt #2866: Web version switcher not working

nico-zhu, 2023-04-18:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The version switcher on the website seems to be not working and only able to stay in version stable https://aedt.docs.pyansys.com/version/stable/index.html# instead of https://aedt.docs.pyansys.com/release/0.6/index.html. 

### Steps To Reproduce

Version dropdown switcher on the website not working. https://aedt.docs.pyansys.com/version/stable/index.html#

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

N/A

MaxJPRey, 2023-04-24:
Hi @nico-zhu 

Can you try: https://aedt.docs.pyansys.com/version/stable/index.html ?

Thanks.

nico-zhu, 2023-04-24:
Hi @MaxJPRey Yes that one works but seems to contain different/less contents than https://aedt.docs.pyansys.com/release/0.6/index.html (eg. core_nets is not included in https://aedt.docs.pyansys.com/version/stable/index.html)

closed 2023-07-19 by Samuelopez-ansys