ansys/pyaedt #2895: Bug located in aedtapp.modeler.primitives[obj_name].faces

jpotter0, 2023-04-21:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I have an HFSS model. I can identify the 3 faces to which I wish to apply H boundary conditions. I do this by using the largest_faces method. I then get a list of all the faces. Then I remove the 3 H boundary faces from the list. Then I use that list to create conducting boundaries by using an HFSS script because the create objects from faces method messes up my frequency calculations due to a "feature" oh HFSS with respect to mesh generation. I will attach screen shots that show the 5 largest faces, the face IDs of the 3 I have selected, a list of all the faces, and the code by which I remove the selected along with the list of the remaining faces.

There are two problems. Face 255 appears in the list of largest faces and is one of my H boundary faces. However, face 255 does not appear in the list of all faces. Iturns out that face 220 is also missing from the list. There are 18 faces in the list, but HFSS shows that there are 20 faces total.

The second problem is that face 29 is not removed. Maybe there is an issue with how I am removing the faces from the list in python. Two o the faces I want to remove are not there to start with and one face I want to remove is not removed. If that's a Python problem, it doesn't relate to the fact that the list of faces is incomplete..

I don't know how to create a simple case to demonstrate this problem.

Here are the screen shots. 

[image]

[image]

[image]

[image]

[image]

[image]

### Steps To Reproduce

1. Get the 5 largest faces
2. Select the 3 faces from that list that correspond to myH boundary faces.
3. Create H boundaries for those faces. All three boundaries are created correctly.
4. Get the list of all faces. 
5. Remove the 3 selected faces.
6. Create conducting boundaries on thee remaining faces.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

[image]

fwlfHS, 2023-04-21:
At least to your 2nd problem: Face 29 is not removed, because `face_id2` contains the id object whereas  `face.id` returns an integer. The comparison will always fail. So the location where you define `face_id2` should be adjusted.

jpotter0, 2023-04-21:
For whatever reason fixing the   face_id2 = faces_largest[3].id statement to get the face id number insteadd f thye face id object fixed both issues. Thanks!

closed 2023-04-21 by jpotter0