ansys/pyaedt #2930: ImportError: No module named skrf

pvinod13, 2023-04-28:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

skrf package is imported as rf for CPython alone not for IPython. However, class TouchstoneData(rf.Network) has an input dependency of skrf.

### Steps To Reproduce

 1. Open AEDT console
 2. from pyaedt import downloads
 3.  from pyaedt.generic.touchstone_parser import read_touchstone
 

UnboundNameException: global name 'rf' is not defined

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ElectronicsDesktop 2023.1.0
IronPython 2.7.0.40 on .NET 4.0.30319.42000

maxcapodi78, 2023-05-02:
@pvinod13 that class is not supposed to work with Ironpyhon, we can add a warning and set rf to none. Do you agree?

MaxJPRey, 2023-05-02:
@maxcapodi78 That would be good indeed to prevent any call from IronPython.

pvinod13, 2023-05-03:
@maxcapodi78, I totally agree with you. But is there any workaround to read the touchstone file in IPython.

closed 2023-10-25 by Samuelopez-ansys