ansys/pyaedt #2942: Student Version of AEDT not working with PyAEDT

RaisinO, 2023-05-02:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I started this in the Q&A section but thought maybe I should post here instead.

I've been struggling to fix an error that has come up after building a Spyder PyAEDT environment in Anaconda and am hoping that someone can point me in the right direction to get things up and running.

I am using the student version of AEDT on a Windows 10 workstation and noticed that when the version of AEDT is called out in some examples, it wasn't clear how to populate the field for the student version as the field usually says '2023.1' while the student version has a different path/version that does not match this format. This is just a guess on my part as to what may be the root of the issue since the error seems to be related to file pathing.

### Steps To Reproduce

Running the following, as suggested in another thread, before running an example:
```
import pyaedt
pyaedt.version
'0.6.73'
```
Followed by:
```
import pyaedt

app = pyaedt.Hfss()
app.release_desktop()
```
Yields the error that I was receiving when I would try to run an example, which is as follows:
```
pyaedt INFO: using existing logger.
pyaedt INFO: AEDT installation Path C:\Program Files\AnsysEM\Ansys Student\v222\Win64
pyaedt INFO: Launching AEDT with the gRPC plugin.
pyaedt INFO: New AEDT session is starting on gRPC port 51737
Exception in thread Thread-6 (launch_desktop_on_port):
Traceback (most recent call last):
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\site-packages\pyaedt\desktop.py", line 79, in launch_desktop_on_port
    subprocess.Popen(" ".join(command), env=my_env, stdout=subprocess.DEVNULL, stderr=subprocess.PIPE)
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 109, in __init__
    super(SubprocessPopen, self).__init__(*args, **kwargs)
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\subprocess.py", line 1440, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
Exception in thread Thread-7 (launch_desktop_on_port):
Traceback (most recent call last):
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\site-packages\pyaedt\desktop.py", line 79, in launch_desktop_on_port
    subprocess.Popen(" ".join(command), env=my_env, stdout=subprocess.DEVNULL, stderr=subprocess.PIPE)
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 109, in __init__
    super(SubprocessPopen, self).__init__(*args, **kwargs)
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\anaconda3\envs\spyder-pyaedt-env\lib\subprocess.py", line 1440, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
pyaedt ERROR: Failed to start AEDT on port 52128.
Traceback (most recent call last):

  Cell In[1], line 3
    app = pyaedt.Hfss()

[... 228 lines cut ...]
toml @ file:///home/conda/feedstock_root/build_artifacts/toml_1604308577558/work
tomli @ file:///home/conda/feedstock_root/build_artifacts/tomli_1644342247877/work
tomlkit @ file:///home/conda/feedstock_root/build_artifacts/tomlkit_1679924068997/work
tornado @ file:///D:/bld/tornado_1681817664954/work
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1675110562325/work
typing_extensions @ file:///home/conda/feedstock_root/build_artifacts/typing_extensions_1678559861143/work
tzdata @ file:///home/conda/feedstock_root/build_artifacts/python-tzdata_1680081134351/work
ujson @ file:///D:/bld/ujson_1675192160047/work
unicodedata2 @ file:///D:/bld/unicodedata2_1667239969015/work
Unidecode @ file:///home/conda/feedstock_root/build_artifacts/unidecode_1664588179651/work
urllib3 @ file:///home/conda/feedstock_root/build_artifacts/urllib3_1678635778344/work
vtk==9.2.6
watchdog @ file:///D:/bld/watchdog_1679333145497/work
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1673864653149/work
webencodings==0.5.1
websocket-client @ file:///home/conda/feedstock_root/build_artifacts/websocket-client_1675567828044/work
whatthepatch @ file:///home/conda/feedstock_root/build_artifacts/whatthepatch_1675090462655/work
widgetsnbextension @ file:///home/conda/feedstock_root/build_artifacts/widgetsnbextension_1680021576815/work
win-inet-pton @ file:///D:/bld/win_inet_pton_1667051142467/work
win32-setctime @ file:///home/conda/feedstock_root/build_artifacts/win32_setctime_1642883564726/work
wrapt @ file:///D:/bld/wrapt_1677485718477/work
wslink @ file:///home/conda/feedstock_root/build_artifacts/wslink_1676499749829/work
yapf @ file:///home/conda/feedstock_root/build_artifacts/yapf_1641487982943/work
yarl @ file:///D:/bld/yarl_1682426748506/work
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1677313463193/work

maxcapodi78, 2023-05-02:
@RaisinO  which Student version have you installed? 
have you tried rebooting machine to let environment variables being recognized after AEDT SV is installed?
Can you check that in your environment variables there is one named ANSYSEMSV_ROOT_xxx ?

RaisinO, 2023-05-02:
@maxcapodi78 Thanks for the reply.
The student version installed is Ansys Electronics Desktop Student 2022 R2
A reboot has been performed after the installation of AEDT SV and after adding "C:\Program Files\AnsysEM\Ansys Student\v222\Win64" manually to the Windows PATH variable
Under environment variables there is ANSYSEMSV_ROOT222 with the correct install path of "C:\Program Files\AnsysEM\Ansys Student\v222\Win64"

maxcapodi78, 2023-05-04:
@RaisinO I think I've figured out the issue. Give it a try in next Pyaedt Version 0.6.74 that I'll publish later this week

RaisinO, 2023-05-06:
@maxcapodi78 It looks like the issue is fixed! I had to reset the Spyder IDE before it would work with the library correctly and call out the student version. Here is the result from the Spyder IPython console:
```
import pyaedt

app = pyaedt.Hfss(student_version=True, non_graphical=True)
app.release_desktop()
pyaedt INFO: using existing logger.
pyaedt INFO: Launching PyAEDT outside AEDT with CPython and PythonNET.
pyaedt INFO: AEDT installation Path C:\Program Files\AnsysEM\Ansys Student\v222\Win64.
pyaedt INFO: Launching AEDT with module PythonNET.
pyaedt INFO: AEDT 2022.2SV Student version started with process ID 30524.
pyaedt INFO: pyaedt v0.6.74
pyaedt INFO: Python version 3.10.10 | packaged by conda-forge | (main, Mar 24 2023, 20:00:38) [MSC v.1934 64 bit (AMD64)]
pyaedt INFO: AEDT 2022.2.0 Build Date 2022-05-26 22:06:26
pyaedt INFO: No project is defined. Project Project4 exists and has been read.
pyaedt INFO: No design is present. Inserting a new design.
pyaedt INFO: Added design 'HFSS_NWL' of type HFSS.
pyaedt INFO: Aedt Objects initialized
pyaedt INFO: Variable Manager initialized
pyaedt INFO: Design Loaded
pyaedt INFO: Materials Loaded
pyaedt INFO: Failed in Adding Desktop Message
pyaedt INFO: Project files removed from handlers.
Out[1]: True
```
Thanks so much.

MaxJPRey, 2023-05-07:
Thanks @RaisinO for the confirmation that the bug was fixed.

closed 2023-05-07 by MaxJPRey