ansys/pyaedt #2944: HFSS Set Solution Type Options

MaxJPRey, 2023-05-02:
### Discussed in https://github.com/pyansys/pyaedt/discussions/2927

<div type='discussions-op-text'>

<sup>Originally posted by **rjwilliams84** April 27, 2023</sup>
Probably a simple solution but I can't find the answer in the documentation or examples.

I'm trying to set up HFSS and define the solution type as 'HFSS with Hybrid and Arrays', 'Modal' solution, with the 'Composite Excitation' option. All I can seem to do is change either 'Modal' or 'Terminal' using the following:

hfss=pyaedt.Hfss(projectname=project_name, designname=design_name ,solution_type="Modal")

where I've tried changing "Modal" after solution_type to other strings, such as "Terminal" (this works fine), "HFSS with Hybrid and Arrays" (does not do anything), and "Composite Excitation" (also does not do anything).

This leads me to believe that I'm not using the correct method to set these solution types/options, but as I mentioned in the beginning of this post I cannot find how to do this in the documentation or in any of the provided examples. I'm hoping someone can point me to what I need to do here.

Thanks.</div>

Samuelopez-ansys, 2023-05-11:
Hi @MaxJPRey ,

You need to enable the "hybrid" property: 
app = Hfss()
app.hybrid = True

It is a bit hidden in the documentation: 
https://aedt.docs.pyansys.com/version/stable/API/_autosummary/pyaedt.hfss.Hfss.hybrid.html
Please if you find a better place to put in the documentation, create it :)

Close this issue if it solved your request. Thank you!

MaxJPRey, 2023-05-11:
@Samuelopez-ansys I agree. I opened it, just to be sure I do not forget about updating the documentation.
I still must do it though.

PR #3085 (2023-06-05): Add documentation for the hybrid solution.
Fix #2944 .
closed 2023-06-01 by svandenb-dev