ansys/pyaedt #2965: build_simulation_project related enhancement

vishak-venkatraman, 2023-05-04:
### Description of the feature

Please enhance build_simulation_project with some switches 

1. A way to turn on/off port generation (needed when we do more custom ports)
2. A way to turn on/off frequency sweep (needed if we do our own custom sweep with hybrid sweep parameters 
3. A way to turn on/off cutout but include net selection. (for example create aedb with only nets of interest and no cut)
4. Able to set hybrid frequency sweep setup as shown below 
```
setup = h3d.create_setup(setupname="Pyaedt_setup",SingleFrequencyDataList__AdaptiveFrequency="1GHZ")
sweep = setup.add_sweep('Sweep1')
sweep.change_type("Interpolating")
sweep.change_range("LinearStep", 100, 1000 , 10, "MHz")
sweep.add_subrange("LinearCount", 0, 10, 1, "Hz")
sweep.add_subrange("LogScale", 10, 1E8, 100, "Hz")
sweep.props["UseQ3DForDC"] = True
```

Samuelopez-ansys, 2023-05-12:
@svandenb-dev I think it is related to Edb

svandenb-dev, 2023-05-31:
@Samuelopez-ansys
 @vishak-venkatraman as we spoke last time I will add more flexibility in the class, I fully agree.

svandenb-dev, 2023-05-31:
PR submitted, closing this issue
@vishak-venkatraman

closed 2023-05-31 by svandenb-dev