ansys/pyaedt #2971: edb.cutout() does not support "clip at extents"

TobiNakamura, 2023-05-05:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

edb.cuout() does not have "clip at extents" feature with custom_extents=True.
If custom extent polygon intersects a net listed in signal_list, it will delete the whole net.

### Steps To Reproduce

Using 09_Configuration.py. Insert below above line 126 (edbapp.nets.plot()):
custom_extent = [[0, 0.045],[0.06,0.045],[0.06,0.02],[0,0.02]]
edbapp.cutout(signal_list=["PCIE0_RX0_P", "PCIE0_RX0_N", "PCIE0_TX0_P_C", "PCIE0_TX0_N_C","PCIE0_TX0_P", "PCIE0_TX0_N"], reference_list=["GND"], custom_extent=custom_extent, custom_extent_units='m')

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

aiofiles==22.1.0
aiosqlite==0.19.0
alabaster==0.7.12
anyio==3.6.2
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
arrow==1.2.3
atomicwrites==1.4.0
attrs==23.1.0
Babel==2.12.1
backcall==0.2.0
beautifulsoup4==4.12.2
bleach==6.0.0
cached-property==1.5.2
certifi==2022.12.7
cffi==1.15.1
chardet==3.0.4
charset-normalizer==3.1.0
cloudpickle==2.2.1
clr-loader==0.2.5
colorama==0.4.6
cycler==0.11.0
Cython==0.29.28
debugpy==1.6.7
decorator==5.1.1
defusedxml==0.7.1
docutils==0.17.1
entrypoints==0.4
execnet==1.9.0
fastjsonschema==2.16.3
fqdn==1.5.1
future==0.16.0
grpcio==1.26.0
hypothesis==6.41.0
idna==3.4
imageio==2.28.0
imagesize==1.3.0
importlib-metadata==6.6.0
importlib-resources==5.12.0
iniconfig==1.1.1
ipycanvas==0.13.1
ipyevents==2.0.1
ipykernel==6.16.2
ipython==7.34.0
ipython-genutils==0.2.0
ipyvtklink==0.2.3
ipywidgets==7.7.5
isoduration==20.11.0
jedi==0.18.2
Jinja2==3.1.2
json5==0.9.11
jsonpointer==2.3
jsonschema==4.17.3
jupyter-events==0.6.3
jupyter-server==1.24.0
jupyter-ydoc==0.2.4
jupyter_client==7.4.9
jupyter_core==4.12.0
jupyter_server_fileid==0.9.0
jupyter_server_ydoc==0.8.0
jupyterlab==3.6.3
jupyterlab-pygments==0.2.2
jupyterlab-widgets==1.1.4
jupyterlab_server==2.22.1
kiwisolver==1.1.0
MarkupSafe==2.1.2
matplotlib==3.1.1
matplotlib-inline==0.1.6
mistune==2.0.5
nbclassic==0.5.6
nbclient==0.7.4
nbconvert==7.3.1
nbformat==5.8.0
nest-asyncio==1.5.6
notebook==6.5.4
notebook_shim==0.2.3
numpy==1.21.5
packaging==23.1
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.5.0
pkgutil_resolve_name==1.3.10
platformdirs==3.5.0
pluggy==1.0.0
plumbum==1.8.1
pooch==1.7.0
prometheus-client==0.16.0
prompt-toolkit==3.0.38
protobuf==3.18.1
psutil==5.9.5
py==1.11.0
pyaedt==0.6.71
pybind11==2.9.2
pycparser==2.21
Pygments==2.15.1
pyparsing==3.0.7
pyrsistent==0.19.3
pytest==7.1.1
pytest-forked==1.4.0
pytest-xdist==2.5.0
python-certifi-win32==1.6
python-dateutil==2.8.2
python-json-logger==2.0.7
pythonnet==3.0.1
pytz==2023.3
pyvista==0.38.5
pywin32==306
pywinpty==2.0.10
PyYAML==6.0
pyzmq==25.0.2
requests==2.29.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpyc==5.3.0
scipy==1.7.0
SCons==4.0.1
scooby==0.7.1
Send2Trash==1.8.2
setuptools-scm==6.4.2
sip==6.5.1
six==1.16.0
sniffio==1.3.0
snowballstemmer==2.2.0
sortedcontainers==2.4.0
soupsieve==2.4.1
Sphinx==4.5.0
sphinxcontrib-applehelp==1.0.2
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==2.0.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.5
spyder-kernels==2.4.3
terminado==0.17.1
tinycss2==1.2.1
toml==0.10.2
tomli==2.0.1
tornado==6.2
traitlets==5.9.0
typing_extensions==4.5.0
uri-template==1.2.0
urllib3==1.26.15
vtk==9.2.6
wcwidth==0.2.6
webcolors==1.13
webencodings==0.5.1
websocket-client==1.5.1
widgetsnbextension==3.6.4
wrapt==1.14.0
y-py==0.5.9
ypy-websocket==0.8.2
zipp==3.15.0

svandenb-dev, 2023-05-08:
@TobiNakamura thank you for your feedback we will check and get back to you.

maxcapodi78, 2023-05-08:
@TobiNakamura  this is not a bug, you used 'm' as unit. you should use 'meter'.

```
edbapp.cutout(signal_list=["PCIE0_RX0_P", "PCIE0_RX0_N", "PCIE0_TX0_P_C", "PCIE0_TX0_N_C","PCIE0_TX0_P", "PCIE0_TX0_N"], reference_list=["GND"], custom_extent=custom_extent, custom_extent_units='meter')

```

closed 2023-05-08 by TobiNakamura