ansys/pyaedt #2978: Hfss :: How to get edge object from selection

bakroistvan, 2023-05-08:
First I select some edge.
```python
app = Hfss()
modeler = app.modeler
selections = modeler.selections

selections[0]
```

This gets me a 'Edge123123' string.
 - Is there a built-in function to export the integer ID from this string?
 - Is there a better way to get the edge object, rather than iterating like in #2367 ?

Samuelopez-ansys, 2023-05-11:
Hi @bakroistvan ,

For the first questions, there is not built-in functions, but with regular expressions, you can quickly get the integer:

```
import re

string = 'Edge123123'

# Use regular expression to match 'Edge' followed by digits
matches = re.findall(r'Edge(\d+)', string)

if matches:
    # Convert the first match to an integer
    number = int(matches[0])
```

Once you have the integer:

1. Get the object name

`object_name = app.modeler.get_object_name_from_edge_id(number)

`2. Get the object edge IDs:
`edges_ids = app.modeler.get_object_edges(object_name)`

3. Get the index of the interested edge:
`edge_index = edges_ids.index(number )`

4. Get the edge object:
`edge_obj = app.modeler[object_name].edges[edge_index]`

Please let me know if it fits your needs and close the issue.

Samuelopez-ansys, 2023-05-17:
@bakroistvan Could I close this issue?

closed 2023-05-30 by Samuelopez-ansys