ansys/pyaedt #2998: Restoring a project archive with two designs with Icepak(...) creates a 3rd design in the project

clguenthe, 2023-05-11:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

If I restore a project archive by passing the archive path to the Icepak(...) constructor, it works fine when the archived project contains a single design. However, with a project with two designs such as 
[image]
the command
`ipk = Icepak(specified_version="2022.2", new_desktop_session=True, projectname="C:\\temp\\ProjectWithTwoDesigns.aedtz")`
erroneously restores to a project with three designs:
[image]

When I restore the archive through the native API command, e.g. by
```
desktop = Desktop(specified_version="2022.2", new_desktop_session=True)
desktop.odesktop.RestoreProjectArchive(archive_path, project_path, True, True)
```
the project is restored correctly.

### Steps To Reproduce

Create a project with two Icepak designs "Design1" and "Design2". Archive the project. Use the command 
`ipk = Icepak(specified_version="2022.2", new_desktop_session=True, projectname="C:\\temp\\ProjectWithTwoDesigns.aedtz")`

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

ansys-templates==0.5.2
arrow==1.2.3
astroid==2.15.1
attrs==22.2.0
autocommand==2.2.2
binaryornot==0.4.4
bottle==0.12.25
build==0.10.0
certifi==2022.12.7
cffi==1.15.1
cfgv==3.3.1
chardet==5.1.0
charset-normalizer==3.1.0
cheroot==9.0.0
click==7.1.2
clr-loader==0.2.5
colorama==0.4.6
contourpy==1.0.7
cookiecutter==2.1.1
coverage==7.2.3
cycler==0.11.0
dill==0.3.6
distlib==0.3.6
exceptiongroup==1.1.1
filelock==3.10.0
flake8==6.0.0
fonttools==4.39.2
gevent==22.10.2
greenlet==2.0.2
identify==2.5.21
idna==3.4
imageio==2.26.1
importlib-metadata==6.1.0
importlib-resources==5.12.0
inflect==6.0.2
iniconfig==2.0.0
isort==5.12.0
jaraco.collections==4.0.0
jaraco.context==4.3.0
jaraco.functools==3.6.0
jaraco.text==3.11.1
Jinja2==3.1.2
jinja2-time==0.2.0
kiwisolver==1.4.4
lazy-object-proxy==1.9.0
MarkupSafe==2.1.2
matplotlib==3.7.1
mccabe==0.7.0
more-itertools==9.1.0
mypy==1.1.1
mypy-extensions==1.0.0
nodeenv==1.7.0
numpy==1.24.2
packaging==23.0
Pillow==9.4.0
platformdirs==3.1.1
pluggy==1.0.0
plumbum==1.8.1
pooch==1.7.0
portend==3.1.0
pre-commit==3.2.0
psutil==5.9.4
py==1.11.0
pyaedt==0.6.61
pycodestyle==2.10.0
pycparser==2.21
pydantic==1.10.7
pyflakes==3.0.1
pylint==2.17.1
pyparsing==3.0.9
pyproject_hooks==1.0.0
pytest==7.2.2
pytest-cov==4.0.0
pytest-html==3.2.0
pytest-metadata==2.0.4
pytest-timeout==2.1.0
pytest-xprocess==0.22.2
python-dateutil==2.8.2
python-slugify==8.0.1
pythonnet==3.0.1
pytz==2023.3
pyvista==0.38.5
pywin32==306
PyYAML==6.0
requests==2.28.2
rpyc==5.3.0
scooby==0.7.1
six==1.16.0
tempora==5.2.1
text-unidecode==1.3
tomli==2.0.1
tomlkit==0.11.7
typing_extensions==4.5.0
urllib3==1.26.15
virtualenv==20.21.0
vtk==9.2.6
waitress==2.1.2
wrapt==1.15.0
zc.lockfile==3.0.post1
zipp==3.15.0
zope.event==4.6
zope.interface==6.0

gmalinve, 2023-05-18:
Hi @clguenthe,

This is correct and has been implemented as a safety reason.
To avoid this behavior you should specify the design name when instantiating Icepack with variable "designname".
Hope this help.

Thanks!

Giulia

sivasu37, 2023-05-19:
Hi @gmalinve , I agree, this is indented behavior. I didn't think of this when I picked up this issue. I will close this issue. 

Thanks

Regards,
Siva

closed 2023-05-19 by sivasu37

clguenthe, 2023-05-19:
One question @gmalinve and @sivasu37: What is the reason for this to happen only to projects with 2 designs, not to projects with a single design? And why is this required for safety reasons? I must say that for me as a user, this behavior is completely unexpected and also not clear through the documentation. Expected behavior from my point of view would be to a) get the same behavior for projects with 1 or 2 designs, and b) have the project "unchanged" through the restore operation. Could you help me here to better understand this? Maybe I have missed an important thought here in the PyAEDT approach?

sivasu37, 2023-05-19:
Hi @clguenthe , When an Icepak object is created it needs to have access to the Icepak design available in the electronics desktop project. 

When you create a Icepak object without any argument to design name it has to look for a available design and load the design in the Icepak object. 

- This is clear when you have only one Icepak design, that design will be loaded into Icepak object. 

- But when you have multiple Icepak designs in the project. We cannot select any one of that design and load into the Icepak object (as it is not clear what design user wants to load). The safer option is to create a new design. If a user wants to load a particular design among the two designs, he has to specify the design name as an argument when creating Icepak objects. In this case, no new Icepak designs will be created.

sivasu37, 2023-05-22:
@clguenthe , Yes, I can add an example showcasing this behavior in the documentation. We currently display the following info in the messages

pyaedt INFO: No consistent unique design is present. Inserting a new design.
pyaedt INFO: Added design 'Icepak_EGA' of type Icepak.

I am also looping in @maxcapodi78 for his comments, As I believe this behavior should be common for other designs like Maxwell and HFSS