ansys/pyaedt #3002: create_touchstone_component freezes program execution

ryand323, 2023-05-12:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

modeler.components.create_touchstone_component() freezes program execution. I am unable to create a Nport component in an aedt circuit.

Program execution continues until this function call, then idles indefinitely.

Not sure if this is a bug, but help is be appreciated.

### Steps To Reproduce

Code:
import os
from matplotlib import pyplot as plt
import numpy as np
import pyaedt

cir = pyaedt.Circuit(projectname=pyaedt.generate_unique_project_name(),
                     new_desktop_session=False,
                     non_graphical=False)

tdr1 = cir.modeler.components.components_catalog["Probes:TDR_Single_Ended"].place("tdr1")

print("Making Touchstone Component")
nport1 = cir.modeler.components.create_touchsthone_component(model_name="dut.s4p") #dut.s4p is in the directory, I have also tried using the full path

Log:
pyaedt INFO: using existing logger.
pyaedt INFO: Launching PyAEDT outside AEDT with CPython and PythonNET.
pyaedt INFO: AEDT installation Path C:\Program Files\AnsysEM\v221\Win64.
pyaedt INFO: Launching AEDT with module PythonNET.
pyaedt INFO: pyaedt v0.6.74
pyaedt INFO: Python version 3.8.8 (default, Apr 13 2021, 15:08:03) [MSC v.1916 64 bit (AMD64)]
pyaedt INFO: AEDT 2022.1.0 Build Date 2021-11-22 22:18:25
pyaedt INFO: Project Project_YUS has been created.
pyaedt INFO: No design is present. Inserting a new design.
pyaedt INFO: Added design 'Circuit Design_XW1' of type Circuit Design.
pyaedt INFO: Aedt Objects initialized
pyaedt INFO: Variable Manager initialized
pyaedt INFO: Design Loaded
pyaedt INFO: Materials Loaded
pyaedt INFO: aedt file load time 0.010759592056274414
Making Touchstone Component

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

alabaster @ file:///home/ktietz/src/ci/alabaster_1611921544520/work
argh==0.26.2
astroid @ file:///C:/b/abs_d4lg3_taxn/croot/astroid_1676904351456/work
asttokens @ file:///opt/conda/conda-bld/asttokens_1646925590279/work
atomicwrites==1.4.0
attrs @ file:///C:/b/abs_09s3y775ra/croot/attrs_1668696195628/work
autopep8 @ file:///tmp/build/80754af9/autopep8_1615918855173/work
Babel @ file:///C:/b/abs_a2shv_3tqi/croot/babel_1671782804377/work
backcall @ file:///home/ktietz/src/ci/backcall_1611930011877/work
bcrypt @ file:///C:/Windows/Temp/abs_36kl66t_aw/croots/recipe/bcrypt_1659554334050/work
beautifulsoup4 @ file:///C:/b/abs_0agyz1wsr4/croot/beautifulsoup4-split_1681493048687/work
black @ file:///C:/b/abs_620t6ndje8/croot/black_1680737261963/work
[... 121 lines cut ...]
sphinxcontrib-jsmath @ file:///home/ktietz/src/ci/sphinxcontrib-jsmath_1611920942228/work
sphinxcontrib-qthelp @ file:///home/ktietz/src/ci/sphinxcontrib-qthelp_1611921055322/work
sphinxcontrib-serializinghtml @ file:///tmp/build/80754af9/sphinxcontrib-serializinghtml_1624451540180/work
spyder @ file:///C:/ci/spyder_1616776239898/work
spyder-kernels @ file:///C:/ci/spyder-kernels_1614030842607/work
stack-data @ file:///opt/conda/conda-bld/stack_data_1646927590127/work
textdistance @ file:///tmp/build/80754af9/textdistance_1612461398012/work
three-merge @ file:///tmp/build/80754af9/three-merge_1607553261110/work
tinycss2 @ file:///C:/b/abs_52w5vfuaax/croot/tinycss2_1668168823131/work
toml @ file:///tmp/build/80754af9/toml_1616166611790/work
tomli @ file:///C:/Windows/TEMP/abs_ac109f85-a7b3-4b4d-bcfd-52622eceddf0hy332ojo/croots/recipe/tomli_1657175513137/work
tomlkit @ file:///C:/Windows/TEMP/abs_3296qo9v6b/croots/recipe/tomlkit_1658946894808/work
tornado @ file:///C:/ci/tornado_1662476991259/work
traitlets @ file:///C:/b/abs_e5m_xjjl94/croot/traitlets_1671143896266/work
typing_extensions @ file:///C:/b/abs_a1bb332wcs/croot/typing_extensions_1681939523095/work
ujson @ file:///C:/ci/ujson_1657525943778/work
urllib3 @ file:///C:/b/abs_3ce53vrdcr/croot/urllib3_1680254693505/work
vtk==9.2.6
watchdog @ file:///C:/ci/watchdog_1612471251191/work
wcwidth @ file:///Users/ktietz/demo/mc3/conda-bld/wcwidth_1629357192024/work
webencodings==0.5.1
win-inet-pton @ file:///C:/ci/win_inet_pton_1605306167264/work
wrapt @ file:///C:/Windows/Temp/abs_7c3dd407-1390-477a-b542-fd15df6a24085_diwiza/croots/recipe/wrapt_1657814452175/work
yapf @ file:///tmp/build/80754af9/yapf_1615749224965/work
zipp @ file:///C:/b/abs_b9jfdr908q/croot/zipp_1672387552360/work

ryand323, 2023-05-18:
Are there any updates to this issue request?

Samuelopez-ansys, 2023-05-19:
Hi @ryand323 ,

Please try to import the NPort manually to verify it is not a problem, of the file.

I have tried with this one:

https://github.com/ansys/pyaedt/blob/main/_unittest/example_models/T21/SSN_ssn.s6p

And it works perfectly:
[image]

ryand323, 2023-05-19:
It does appear to be related to the s-parameter files being imported. SSN_ssn.s6p imported correctly with pyaedt into the circuit simulator. 

Opening up the touchsone files for comparison, it appears SSN_ssn.s6p is an exported file from aedt:

> ! Touchstone file exported from HFSS 3D Layout Design 2021.2.0
> !        File:           D:/Simulation/SSN.aedt
> !        Generated:      12:36:15 PM  Feb 18, 2021
> !        Design:         ssn
> !        Project:        SSN
> !        Setup:          SYZ Sweep 1
> !        Solution:       SYZ Sweep 1
> !
> (Pound Sign Removed for Github Markup) GHz S MA R 50.000000
> ! Terminal data exported
> ! Port[1] = A-MII-RXD1_30_SQFP28X28_208_U1
> ! Port[2] = A-MII-RXD1_65_SQFP20X20_144_U7
> ! Port[3] = A-MII-RXD2_32_SQFP28X28_208_U1
> ! Port[4] = A-MII-RXD2_66_SQFP20X20_144_U7
> ! Port[5] = A-MII-RXD3_33_SQFP28X28_208_U1
> ! Port[6] = A-MII-RXD3_67_SQFP20X20_144_U7
> 0                              0.00298166409530576 0                0.997018335804667 0                9.71569326119175E-06 0                9.71569326119176E-06 180              
>                                9.71553401690422E-06 0                9.71553398912824E-06 180              
>                                0.997018335804667 0                0.00298166409530576 0                9.71569326119175E-06 180              9.71569326119175E-06 0                
>                                9.71553398921538E-06 180              9.71553401684487E-06 0                
>                                9.71569326119175E-06 0                9.71569326119175E-06 180              0.00299134546858548 0                0.997008654431399 0                
>                                9.71543966877399E-06 0                9.71543965339285E-06 180              
>                                9.71569326119176E-06 180              9.71569326119175E-06 0                0.997008654431399 0                0.0029913454685857 0                
>                                9.71543965340107E-06 180              9.71543966877518E-06 0                
>                                9.71553401690422E-06 0                9.71553398921538E-06 180              9.71543966877399E-06 0                9.71543965340107E-06 180              
>                                0.00300768812298791 0                0.996992311774038 0                
>                                9.71553398912824E-06 180              9.71553401684487E-06 0                9.71543965339285E-06 180              9.71543966877518E-06 0                
>                                0.996992311774038 0                0.00300768812300545 0

The s-parameter file I would like to use comes from a VNA measurement. The way data is formatted is quite a bit different, not to mention different units. 

> !Date: Tuesday, April 25, 2023 08:31:13
> !Correction: S11(C  4-Port  ) 
> !S12(C  4-Port  ) 
> !S13(C  4-Port  ) 
> !S14(C  4-Port  ) 
> !S21(C  4-Port  ) 
> !S22(C  4-Port  ) 
> !S23(C  4-Port  ) 
> !S24(C  4-Port  ) 
> !S31(C  4-Port  ) 
> !S32(C  4-Port  ) 
> !S33(C  4-Port  ) 
> !S34(C  4-Port  ) 
> !S41(C  4-Port  ) 
> !S42(C  4-Port  ) 
> !S43(C  4-Port  ) 
> !S44(C  4-Port  ) 
> !S4P File: Measurements: <S11,S12,S13,S14>,
> !<S21,S22,S23,S24>,
> !<S31,S32,S33,S34>,
> !<S41,S42,S43,S44>:
> (Pound Sign Removed for Github Markup) Hz S  dB   R 50
> 10000000 -38.25531 -78.733177 -0.025349155 0.015774187 -63.130737 83.516792 -71.892441 -98.401344
> -0.021411726 0.032946281 -39.054512 -74.961319 -72.006302 -100.13907 -62.459106 81.599861
> -63.774036 84.543793 -73.498238 -101.85165 -38.741035 -78.042351 -0.027126838 0.00080163183
> -74.801346 -98.826355 -63.534279 83.437126 -0.025177831 0.0071976553 -40.113575 -74.683968

Does the pyaedt function only accept files with the AEDT formatting and that is all this issue is?

Samuelopez-ansys, 2023-05-20:
@ryand323 Did you try to import this VNA touchstone in Circuit manually?

Because if it does not work, you should contact the Ansys support to help you because it is related to AEDT.

pyAEDT accepts the same that Aedt accepts, if  in AEDT works please share the file, because one solution could be formatting the file.

ryand323, 2023-05-22:
@Samuelopez-ansys The VNA touchstone s-parameters can be loaded into circuit manually. It does not load with pyaedt. The AEDT and Keysight files are delimited very different, perhaps this is a python problem. See the attached s4p file.
[Keysight_VNA_microstrip.zip](https://github.com/ansys/pyaedt/files/11533668/Keysight_VNA_microstrip.zip)

PR #3057 (2023-05-25): Add new touchstone format. It was working only with Ansys output format "! Port"
Close #3002
closed 2023-05-25 by Samuelopez-ansys