ansys/pyaedt #3013: Delete unused variables

arifroni, 2023-05-16:
Hello

1. How can I delete all the unused design variables from the project? I found [this](https://aedt.docs.pyansys.com/version/stable/API/_autosummary/pyaedt.application.Variables.VariableManager.independent_design_variable_names.html#pyaedt.application.Variables.VariableManager.independent_design_variable_names) function, it only shows the variables without some expression.
2. Is there any way to get a list of things where a variable has been used? eg: geometry, co-ordinate system, etc.

Samuelopez-ansys, 2023-05-19:
Hi @arifroni,

This method is not implemented, we will consider it.

In the meanwhile, you can try to delete all variables in a for loop:

app.variable_manager.delete_variable("Var1")

If the variable is used, it should not have any effect because AEDT does not allow to delete used variables.. Please let me know if it works

Samuelopez-ansys, 2023-09-22:
@maxcapodi78 Maybe you could add the method you developed to delete all unused variables

closed 2023-10-26 by Samuelopez-ansys