ansys/pyaedt #3056: Problem compiling pyaedt application and running in Ansys 2023.1

adamgregory81, 2023-05-25:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The issue is not directly related to pyaedt but comes about when attempting to compile an application using pyaedt with PyInstaller.  My compiled application works with Ansys 2022 versions, but it will not run using Ansys 2023.1

In 2023, this .dll is loaded:
C:\Program Files\AnsysEM\v231\Win64\PyDesktopPlugin.dll

In 2022, the .dll is not loaded.  (Not sure why 2023 needs to load it but not 2022).

When running from python source code, there is no problem loading the .dll.  But after compiling with PyInstaller, I get this error message:
File "C:\Program Files\AnsysEM\v231\Win64\PythonFiles\DesktopPlugin\PyDesktopPluginDll.py", line 33, in <module> 
    AedtAPI = PyDLL(AedtAPIDll_file) 
  File "PyInstaller\loader\pyimod03_ctypes.py", line 66, in __init__ 
pyimod03_ctypes.PyInstallerImportError: Failed to load dynlib/dll 'C:\\Program Files\\AnsysEM\\v231\\Win64\\PyDesktopPlugin.dll'. Most likely this dynlib/dll was not found when the application was frozen. 
[14240] Failed to execute script 'borgen_main' due to unhandled exception!

The .dll does exist at the path listed, so that is not the problem.  I did some searches online of similar error messages, and I managed to find a solution.  Unfortunately I had to edit this file:
C:\Program Files\AnsysEM\v231\Win64\PythonFiles\DesktopPlugin\PyDesktopPluginDll.py

Line 32:
AedtAPI = PyDLL(AedtAPIDll_file)

I changed to:
AedtAPI = PyDLL(AedtAPIDll_file, winmode=0)

Setting the winmode to 0 instead of letting the function determine the winmode seemed to allow the .dll path to be discovered in the python executable.

I would prefer to have a solution that does not require me to edit this file.  Not sure if pyaedt can do anything to assist in the way the DLL is loaded.

### Steps To Reproduce

Follow description above

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

aiohttp==3.8.3
aiosignal==1.2.0
altgraph==0.17.3
appdirs==1.4.4
async-timeout==4.0.2
attrs==22.1.0
cachetools==5.3.0
cffi==1.15.1
charset-normalizer==2.1.1
clr-loader==0.2.3
contourpy==1.0.5
cycler==0.11.0
et-xmlfile==1.1.0
fonttools==4.37.3
frozenlist==1.3.1
idna==3.4
imageio==2.22.0
kiwisolver==1.4.4
matplotlib==3.6.0
multidict==6.0.2
numpy==1.23.3
openpyxl==3.0.10
packaging==21.3
pandas==1.5.1
pefile==2023.2.7
Pillow==9.2.0
plumbum==1.7.2
psutil==5.9.2
py2exe==0.13.0.0
pyaedt==0.6.76
pycparser==2.21
pyinstaller==5.10.1
pyinstaller-hooks-contrib==2023.2
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.2.1
pyvista==0.36.1
pywin32==304
pywin32-ctypes==0.2.0
rpyc==5.3.1
scooby==0.6.0
six==1.16.0
vtk==9.1.0
wslink==1.8.2
yarl==1.8.1

Samuelopez-ansys, 2023-05-26:
Hi @adamgregory81 ,

From PyAEDT side, we can not do much because as you know, it needs this DLL. Actually, we realized this issue when we tried to use CPython 3.7 from AEDT as a Base interpreter. If you check the installation script:
https://aedt.docs.pyansys.com/version/stable/Getting_started/Installation.html#install-from-a-python-file
You can see how you need to uninstall pywin32, for the same reason as you mentioned.

Then, please submit a defect in Ansys internal system, because it is related to the Ansys AEDT core team.

@maxcapodi78

adamgregory81, 2023-05-26:
Okay, I will submit a defect with internal Ansys.

One other note regarding this topic:  When I use Ansys 2023, I see that this logic block is entered within desktop.py
            elif settings.use_grpc_api is None or _com == "gprc_v3":
                settings.use_grpc_api = True

For Ansys 2022, settings.use_grpc_api was set to False.  Setting it to True is what causes the .dll to be accessed.  I don't know anything about what "grpc_api" refers to, but as a test, I set a breakpoint and forced use_grpc_api to be False when using Ansys 2023.  This removed the need to have  the .dll and everything seemed to work properly.  Again I have no idea if that creates other problems, but I thought it was worth noting.

Samuelopez-ansys, 2023-05-29:
@adamgregory81 

It was an internal decision. In 2023R1, GRPC mode was implemented in AEDT (it was beta in 2022R2), it is a new way to connect to AEDT, instead of COM (original one). The advantage of using GRPC is, you can connect remotely and it is more stable, then by default from 2023R1, PyAEDT opens AEDT with GRCP.

The problem is, when you open AEDT manually (with the icon) it is still opening with COM, then if you want to connect to an existing session, you need to change settings.use_grcp_api = False. Or, in 2023R1, a bug in the structure folder was introduced, and in some cases due to the .dll in the main folder, GRPC could fail. This has been fixed in 2023R2, if you can test it.

In summary, you can change to False the pyaedt.settings.use_grcp_api variable before launching AEDT with PyAEDT, but AEDT and PyAEDT future is using GRPC.

closed 2023-05-29 by Samuelopez-ansys