ansys/pyaedt #3068: PyAEDT error creating Wave Port

AndrewKee, 2023-05-30:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

the hfss.wave_port function errors out when adding integration lines
Errors provided
pyaedt ERROR: Error in :   File "pyaedt\hfss.py", line 3029, in _get_reference_and_integration_points
pyaedt ERROR: list.remove(x): x not in list - _get_reference_and_integration_points -  VALUE ERROR.
pyaedt ERROR: Arguments with values: 
pyaedt ERROR:     sheet = port_1_sheet 
pyaedt ERROR:     axisdir = 3 
pyaedt ERROR:     obj_name =  
pyaedt ERROR: Check Online documentation on: https://aedt.docs.pyansys.com/version/stable/search.html?q=_get_reference_and_integration_points+sheet+axisdir+obj_name

Commenting out the line 3029 in hfss.py resolves this issue.  However unclear from an outsider perspective how important this line is.
"refid.remove(sheet)"

### Steps To Reproduce

port_1_sheet = modeler.create_rectangle(pyaedt.constants.AXIS.Z, 
                                                ["-sept_b0/2", "-sept_a0/2", "0"], 
                                                ["sept_b0", "sept_a0"], 
                                                name="port_1_sheet", matname="vacuum", 
                                                is_covered=True)
port1 = hfss.wave_port(port_1_sheet, integration_line=hfss.AxisDir.XPos, name="1", num_modes=1, renormalize=False)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

alabaster @ file:///home/ktietz/src/ci/alabaster_1611921544520/work
anaconda-client==1.11.0
anaconda-navigator==2.3.2
anaconda-project @ file:///C:/Windows/TEMP/abs_91fu4tfkih/croots/recipe/anaconda-project_1660339890874/work
anyio @ file:///C:/ci/anyio_1644481921011/work/dist
appdirs==1.4.4
argon2-cffi @ file:///opt/conda/conda-bld/argon2-cffi_1645000214183/work
argon2-cffi-bindings @ file:///C:/ci/argon2-cffi-bindings_1644551690056/work
arrow @ file:///opt/conda/conda-bld/arrow_1649166651673/work
astroid @ file:///C:/Windows/TEMP/abs_b0dtxgpicv/croots/recipe/astroid_1659023126745/work
atomicwrites==1.4.0
attrs @ file:///opt/conda/conda-bld/attrs_1642510447205/work
Automat @ file:///tmp/build/80754af9/automat_1600298431173/work
autopep8 @ file:///opt/conda/conda-bld/autopep8_1650463822033/work
Babel @ file:///tmp/build/80754af9/babel_1620871417480/work
backcall @ file:///home/ktietz/src/ci/backcall_1611930011877/work
backports.functools-lru-cache @ file:///tmp/build/80754af9/backports.functools_lru_cache_1618170165463/work
backports.tempfile @ file:///home/linux1/recipes/ci/backports.tempfile_1610991236607/work
backports.weakref==1.0.post1
bcrypt @ file:///C:/Windows/Temp/abs_36kl66t_aw/croots/recipe/bcrypt_1659554334050/work
beautifulsoup4 @ file:///C:/ci/beautifulsoup4_1650293025093/work
binaryornot @ file:///tmp/build/80754af9/binaryornot_1617751525010/work
bitarray @ file:///C:/ci/bitarray_1657729621682/work
black @ file:///C:/ci/black_1660239974023/work
bleach @ file:///opt/conda/conda-bld/bleach_1641577558959/work
bokeh @ file:///C:/Windows/TEMP/abs_4a259bc2-ed05-4a1f-808e-ac712cc0900cddqp8sp7/croots/recipe/bokeh_1658136660686/work
boltons @ file:///C:/b/abs_707eo7c09t/croot/boltons_1677628723117/work
boto3 @ file:///C:/Windows/TEMP/abs_4009c406-44ba-4406-8996-204d9b11202flt4kglbk/croots/recipe/boto3_1657820114895/work
botocore @ file:///C:/ci/botocore_1657735875454/work
Bottleneck @ file:///C:/Windows/Temp/abs_3198ca53-903d-42fd-87b4-03e6d03a8381yfwsuve8/croots/recipe/bottleneck_1657175565403/work
brotlipy==0.7.0
[... 260 lines cut ...]
twisted-iocpsupport @ file:///C:/ci/twisted-iocpsupport_1646798932792/work
typing_extensions @ file:///C:/Windows/TEMP/abs_dd2d0moa85/croots/recipe/typing_extensions_1659638831135/work
ujson @ file:///C:/ci/ujson_1657525944442/work
Unidecode @ file:///tmp/build/80754af9/unidecode_1614712377438/work
urllib3 @ file:///C:/Windows/TEMP/abs_65ynz4fdmi/croots/recipe/urllib3_1659110473919/work
vtk==9.2.6
w3lib @ file:///Users/ktietz/demo/mc3/conda-bld/w3lib_1629359764703/work
watchdog @ file:///C:/ci/watchdog_1638367441841/work
wcwidth @ file:///Users/ktietz/demo/mc3/conda-bld/wcwidth_1629357192024/work
webencodings==0.5.1
websocket-client @ file:///C:/ci/websocket-client_1614804375980/work
Werkzeug @ file:///opt/conda/conda-bld/werkzeug_1645628268370/work
widgetsnbextension @ file:///C:/ci/widgetsnbextension_1644991377168/work
win-inet-pton @ file:///C:/ci/win_inet_pton_1605306162074/work
win-unicode-console==0.5
wincertstore==0.2
wrapt @ file:///C:/Windows/Temp/abs_7c3dd407-1390-477a-b542-fd15df6a24085_diwiza/croots/recipe/wrapt_1657814452175/work
xarray==0.20.1
xlrd @ file:///tmp/build/80754af9/xlrd_1608072521494/work
XlsxWriter @ file:///opt/conda/conda-bld/xlsxwriter_1649073856329/work
xlwings @ file:///C:/b/abs_41rmpiyy58/croots/recipe/xlwings_1664932818417/work
yapf @ file:///tmp/build/80754af9/yapf_1615749224965/work
zict==2.1.0
zipp @ file:///C:/ci/zipp_1652273994994/work
zope.interface @ file:///C:/ci/zope.interface_1625036252485/work

Samuelopez-ansys, 2023-05-31:
Hi @AndrewKee,

This is working for me:

```
app = pyaedt.Hfss(new_desktop_session=False,
                  non_graphical=False,
                  specified_version="2023.1")

app["sept_b0"] = "10mm"
app["sept_a0"] = "20mm"

port_1_sheet = app.modeler.create_rectangle(pyaedt.constants.AXIS.Z,
                                        ["-sept_b0/2", "-sept_a0/2", "0"],
                                        ["sept_b0", "sept_a0"],
                                        name="port_1_sheet",
                                        matname="vacuum",
                                        is_covered=True)

port1 = app.wave_port(port_1_sheet, integration_line=app.AxisDir.XPos, name="1", num_modes=1, renormalize=False)
```

app.release_desktop(True, True)

Which AEDT version are you using? What are the parameter values?

Samuelopez-ansys, 2023-06-08:
Hi @AndrewKee any update?

AndrewKee, 2023-06-08:
Apologies.  Was on vacation the past week.  Will do some further tests shortly.  
I'm currently using 2022 R2
The variables are defined as such, where I have a coefficient multiplied by another value.
hfss["sept_a0_coef"] = "0.626"
hfss["sept_a0"]      = "sept_a0_coef*lambda_0"

perhaps an issue with 2022 R2.  I will investigate further.

AndrewKee, 2023-06-12:
The issue does not occur with this simple code you wrote in 2022R2 as well.  Digging deeper into my code to see what else could be causing this issue.

AndrewKee, 2023-06-12:
The issue occurs when you do not have a variable defined... 
in the above comment, lambda_0 was not defined.  Everything was somehow still being produced correctly, yet it was the port creation that errored.

closed 2023-06-12 by AndrewKee