ansys/pyaedt #3148: primitives mess up after import stackup.xml

ring630, 2023-06-19:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

some primitives are moved from one layer to another after stackup.xml import.

**Before loading stackup.xml**
**stackup layer 1_Top 1228**
stackup layer DE1 0
stackup layer Inner1(GND1) 2
stackup layer DE2 0
stackup layer Inner2(PWR1) 5
stackup layer DE3 0
stackup layer Inner3(Sig1) 114
stackup layer Megtron4-1mm 0
stackup layer Inner4(Sig2) 112
stackup layer DE5 0
stackup layer Inner5(PWR2) 4
stackup layer DE6 0
stackup layer Inner6(GND2) 1
stackup layer DE7 0
stackup layer 16_Bottom 349
stackup layer Top Overlay 195
stackup layer Top Solder 24
stackup layer Bottom Solder 6
stackup layer Bottom Overlay 66
stackup layer Outline 1

**After loading stackup.xml**
**stackup layer 1_Top 0**
stackup layer DE1 0
stackup layer Inner1(GND1) 1228
stackup layer DE2 0
stackup layer Inner2(PWR1) 2
stackup layer DE3 0
stackup layer Inner3(Sig1) 5
stackup layer Megtron4-1mm 0
stackup layer Inner4(Sig2) 114
stackup layer DE5 0
stackup layer Inner5(PWR2) 112
stackup layer DE6 0
stackup layer Inner6(GND2) 4
stackup layer DE7 0
stackup layer 16_Bottom 350
stackup layer Top Overlay 195
stackup layer Top Solder 24
stackup layer Bottom Solder 6
stackup layer Bottom Overlay 66
stackup layer Outline 1
***

### Steps To Reproduce

import pyaedt
from pyaedt import Edb
from pathlib import Path

def get_ansys_pcb():
    temppath = "edb_demo"
    fname = "ANSYS-HSD_V1.aedb"

    fpath = Path(temppath) / "edb" / 'ANSYS-HSD_V1.aedb'
    if not fpath.exists():
        fpath = pyaedt.downloads.download_file(f'edb/{fname}', destination=temppath)
    return str(fpath)

fpath = get_ansys_pcb()

appedb = Edb(fpath, edbversion="2023.1")
print("Before loading stackup.xml")
for i, j in appedb.stackup.layers.items():
    print(f"stackup layer {i} {len(appedb.modeler.get_primitives(layer_name=i))}")

print("")
appedb.stackup.export("stk.xml")
appedb.stackup.load("stk.xml")

print("After loading stackup.xml")
for i, j in appedb.stackup.layers.items():
    print(f"stackup layer {i} {len(appedb.modeler.get_primitives(layer_name=i))}")

print("***")

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

all

svandenb-dev, 2023-06-28:
@ring630 what it the status ? if you need any help feel free to contact me.

ring630, 2023-06-28:
@svandenb-dev It is a bug in EDB API. I will find a workaround to fix it.

ring630, 2023-07-14:
Rootcause is found. After loading xml, the layer id got reset. layer collection needs to be adjusted before setting into stackup.

closed 2023-07-17 by ring630