ansys/pyaedt #3182: edb.cutout() clip at extent keep path as path

TobiNakamura, 2023-06-26:
### Description of the feature

Currently when pyaedt cutout is performed using a custom extent which intersects a path, the path is converted to a polygon as a consequence of the subtraction operation. Instead, the cutout should maintain the clipped path as a path. See cutout using use_pyaedt_cutout=False as an example.

svandenb-dev, 2023-06-28:
hi @TobiNakamura having path converted into polygon is the "normal" behavior from EDB native API. There is not so much we can do. You can also get the PoygonData from a Path and treat it as a polygon too. I assume you want to access the center_line and width from path after clipping ?

svandenb-dev, 2023-06-28:
Closing this isse since it's due to native API behavior and also solution for creating ports on polygons was provided

closed 2023-06-28 by svandenb-dev

TobiNakamura, 2023-06-28:
When you cutout using use_pyaedt_cutout=False OR cutout through the desktop app, clipped paths remain as paths. This issue seems specific to the pyaedt implementation of cutout.

I would like center_line and width after clipping. Could you show me how to access this? 
project.nets.nets['PE_RXN0'].primitives[1].polygon_data does not have the relevant parameters.
project.nets.nets['PE_RXN0'].primitives[1].path.width returns:
Traceback (most recent call last):
  File "c:\Users\tshiresn\AppData\Roaming\pyaedt_env_ide\v222\lib\site-packages\pyaedt\edb_core\dotnet\primitive.py", line 36, in __getattr__
    return super().__getattribute__(key)
  File "c:\Users\tshiresn\AppData\Roaming\pyaedt_env_ide\v222\lib\site-packages\pyaedt\edb_core\dotnet\primitive.py", line 799, in width
    return self.prim_obj.GetWidth()
AttributeError: 'NoneType' object has no attribute 'GetWidth'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\tshiresn\AppData\Roaming\pyaedt_env_ide\v222\lib\site-packages\pyaedt\edb_core\dotnet\primitive.py", line 43, in __getattr__
    return getattr(obj, key)
AttributeError: width

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "c:\Users\tshiresn\AppData\Roaming\pyaedt_env_ide\v222\lib\site-packages\pyaedt\edb_core\dotnet\primitive.py", line 45, in __getattr__
    raise AttributeError("Attribute not present")
AttributeError: Attribute not present