ansys/pyaedt #3183: Bug located in pyAEDT create_bondwire

enriquemarquez, 2023-06-26:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The create_bondwire do not allow str as input arguments. create_box accept it. Additionally using pyAEDT it is not possible to change orientation axis to Y or X. The error message given is:

pyaedt INFO: using existing logger.
pyaedt INFO: Launching PyAEDT outside AEDT with CPython and PythonNET.
pyaedt INFO: AEDT installation Path C:\Program Files\AnsysEM\v222\Win64.
pyaedt INFO: Launching AEDT with module PythonNET.
pyaedt INFO: Ansoft.ElectronicsDesktop.2022.2 Started with process ID 11168.
pyaedt INFO: pyaedt v0.6.45
pyaedt INFO: Python version 3.10.5 (tags/v3.10.5:f377153, Jun  6 2022, 16:14:13) [MSC v.1929 64 bit (AMD64)]
pyaedt INFO: No project is defined. Project Project5 exists and has been read.
pyaedt INFO: Active Design set to HFSS_GC7
pyaedt INFO: Aedt Objects initialized
pyaedt INFO: Variable Manager initialized
pyaedt INFO: Design Loaded
pyaedt INFO: Successfully loaded project materials !
pyaedt INFO: Materials Loaded
pyaedt ERROR: Error in :   File "C:\Users\erago\AppData\Local\Programs\Python\Python310\lib\site-packages\pyaedt\modeler\cad\Primitives3D.py", line 520, in create_bondwire
pyaedt ERROR: unsupported operand type(s) for -: 'str' and 'str' - create_bondwire -  TYPE ERROR.
pyaedt ERROR: Arguments with values:
pyaedt ERROR:     start_position = ['-$msWinA + $msWf/2 +2  * 0 * ($msWf + $msSf)', '$substrateThickness + $msT', '$msLin + $msLinA +$msSin + $msLf - $bondSep']
pyaedt ERROR:     end_position = ['-$msWinA + $msWf/2 +2 *0 *($msWf + $msSf) + 2 * ($msSf + $msWf)', '$substrateThickness + $msT', '$msLin + $msLinA +$msSin + $msLf - $bondSep']  
pyaedt ERROR:     h1 = $bondHeight
pyaedt ERROR:     h2 = 0mm
pyaedt ERROR:     alpha = None
pyaedt ERROR:     beta = None
pyaedt ERROR:     bond_type = 1
pyaedt ERROR:     diameter = 0.025
pyaedt ERROR:     facets = 6
pyaedt ERROR:     name = Bond
pyaedt ERROR:     matname = pec
pyaedt ERROR: Check Online documentation on: https://aedt.docs.pyansys.com/search.html?q=create_bondwire
pyaedt ERROR: Error in :   File "C:\Users\erago\AppData\Local\Programs\Python\Python310\lib\site-packages\pyaedt\modeler\cad\Primitives3D.py", line 520, in create_bondwire        
pyaedt ERROR: unsupported operand type(s) for -: 'str' and 'str' - create_bondwire -  TYPE ERROR.
pyaedt ERROR: Arguments with values:
pyaedt ERROR:     start_position = ['-$msWinA + $msWf/2 +2  * 1 * ($msWf + $msSf)', '$substrateThickness + $msT', '$msLin + $msLinA +$msSin + $msLf - $bondSep']
pyaedt ERROR:     end_position = ['-$msWinA + $msWf/2 +2 *1 *($msWf + $msSf) + 2 * ($msSf + $msWf)', '$substrateThickness + $msT', '$msLin + $msLinA +$msSin + $msLf - $bondSep']  
pyaedt ERROR:     h1 = $bondHeight
pyaedt ERROR:     h2 = 0mm
pyaedt ERROR:     alpha = None
pyaedt ERROR:     beta = None
pyaedt ERROR:     bond_type = 1
pyaedt ERROR:     diameter = 0.025
pyaedt ERROR:     facets = 6
pyaedt ERROR:     name = Bond
pyaedt ERROR:     matname = pec
pyaedt ERROR: Check Online documentation on: https://aedt.docs.pyansys.com/search.html?q=create_bondwire

### Steps To Reproduce

import math, cmath

from pyaedt import Hfss
hfss= Hfss()
hfss.modeler.model_units ="mm"

## Variables
Ni = 3
No = 3
hfss["$msNfin"] = str(Ni)
hfss["$msNfout"] = str(No)
hfss["$msLin"] = "3.0mm"
hfss["$msWin"] = "1.62mm"
hfss["$msWf"] = "0.170mm"    
hfss["$msSf"] = "0.120mm"    
hfss["$msSin"] = "0.120mm"   
hfss["$msLf"] = "9.5mm"    
hfss["$msLinA"] = "1.0mm"    
hfss["$msT"] = "0.032mm"    

hfss['$msWinA'] = "($msNfin + $msNfout) * $msWf + ($msNfin + $msNfout -1) * $msSf"

hfss["$substrateThickness"] = "30mils"   
hfss["$substrateWidth"]     = "$msWinA * 7"  
hfss["$substrateLength"]    = "2 * $msLin + 2 * $msLinA + $msLf + $msSin" 
         
hfss["$bondDiameter"] = "0.025mm"
hfss["$bondHeight"] = "0.1mm"
hfss["$bondSep"] = "0.15mm"
## Create fingers
for i in range(0,Ni):
    fi=hfss.modeler.create_box(["-$msWinA/2 + 2 *" +  str(i) +" * ($msWf + $msSf)","$substrateThickness","$msLin + $msLinA"],
                                    ["$msWf","$msT", "$msLf"], 
                                     name="fingerIn", matname="pec")
# Create Wire bondings
for i in range(0,Ni-1):
    origin = ["-$msWinA + $msWf/2 +2  * "+ str(i) + " * ($msWf + $msSf)", "$substrateThickness + $msT", "$msLin + $msLinA +$msSin + $msLf - $bondSep"]
    endpos = ["-$msWinA + $msWf/2 +2 *"+ str(i) + " *($msWf + $msSf) + 2 * ($msSf + $msWf)", "$substrateThickness + $msT", "$msLin + $msLinA +$msSin + $msLf - $bondSep"]
    bi = hfss.modeler.create_bondwire(origin, endpos,
                                   	h1="$bondHeight", h2="0mm", 
                                   	bond_type=1, diameter = "$bondDiameter", facets = 6,
                                    name="Bond", matname="pec")

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

absl-py==1.1.0
appdirs==1.4.4
asttokens==2.2.1
astunparse==1.6.3
backcall==0.2.0
cachetools==5.2.0
certifi==2022.6.15
cffi==1.15.1
charset-normalizer==2.0.12
clr-loader==0.2.5
colorama==0.4.6
comm==0.1.2
cycler==0.11.0
debugpy==1.6.5
decorator==5.1.1
entrypoints==0.4
executing==1.2.0
flatbuffers==1.12
fonttools==4.33.3
fpdf==1.7.2
gast==0.4.0
gmsh==4.11.1
google-auth==2.8.0
google-auth-oauthlib==0.4.6
google-pasta==0.2.0
grpcio==1.47.0
h5py==3.7.0
hycohanz @ https://github.com/Pablo097/hycohanz/archive/devel.zip
idna==3.3
imageio==2.24.0
ipykernel==6.20.1
ipython==8.8.0
jedi==0.18.2
jupyter_client==7.4.8
jupyter_core==5.1.3
keras==2.9.0
Keras-Preprocessing==1.1.2
kiwisolver==1.4.3
libclang==14.0.1
Markdown==3.3.7
matplotlib==3.5.2
matplotlib-inline==0.1.6
nest-asyncio==1.5.6
numpy==1.23.0
oauthlib==3.2.0
opt-einsum==3.3.0
packaging==21.3
pandas==1.4.3
parso==0.8.3
pickleshare==0.7.5
Pillow==9.1.1
platformdirs==2.6.2
plumbum==1.8.1
pooch==1.6.0
progress==1.6
progress-bar==8
progressbar==2.5
progressbar2==4.0.0
prompt-toolkit==3.0.36
protobuf==3.19.4
psutil==5.9.4
pure-eval==0.2.2
pyaedt==0.6.45
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycparser==2.21
Pygments==2.14.0
pyparsing==3.0.9
python-dateutil==2.8.2
python-utils==3.3.3
pythonnet==3.0.1
pytz==2022.1
pyvista==0.37.0
pywin32==305
pyzmq==24.0.1
quantiphy==2.19
requests==2.28.0
requests-oauthlib==1.3.1
rpyc==5.3.0
rsa==4.8
scikit-rf==0.24.1
scipy==1.10.0
scooby==0.7.0
six==1.16.0
stack-data==0.6.2
tensorboard==2.9.1
tensorboard-data-server==0.6.1
tensorboard-plugin-wit==1.8.1
tensorflow==2.9.1
tensorflow-estimator==2.9.0
tensorflow-io-gcs-filesystem==0.26.0
termcolor==1.1.0
tornado==6.2
traitlets==5.8.1
typing_extensions==4.2.0
urllib3==1.26.9
vtk==9.2.5
wcwidth==0.2.5
Werkzeug==2.1.2
wrapt==1.14.1

gmalinve, 2023-06-29:
Hi @enriquemarquez ,

We are fixing the bug so we can now pass parameters as inputs.
Feature will be available in next release.

Thanks for reporting the issue!

Giulia

gmalinve, 2023-06-29:
@enriquemarquez in the meanwhile you can test the branch fix/issue_3183 #3203 .
In case you find something else please report it here.

Thanks!

closed 2023-06-29 by gmalinve