ansys/pyaedt #3184: 2021.2 Hfss instance not working

arobinson905, 2023-06-26:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

2021.2 the call Hfss() is not initiated. 

### Steps To Reproduce

With a 2021.2 project open, try performing any pyansys script on this desktop instance. Using Hfss() in the pyaedt kit will throw non graphical and desktop initiation errors. This is true for 3.11, 3.10, and 3.9 python versions. Also the python version ask in this form only goes up to 3.10
project_name = "C:\\Users\\arobinso\\OneDrive - ANSYS, Inc\\Documents\\CUBoulder\\Bowtie_T.aedt"
design_name = "Skin"
non_graphical = os.getenv("PYAEDT_NON_GRAPHICAL", "False").lower() in ("true", "1", "t")
aedtapp = Hfss(specified_version="2021.2", projectname=project_name, designname=design_name, non_graphical=non_graphical, new_desktop_session=False)

Traceback (most recent call last):
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\CUBoulder\pyaedt_script.py", line 11, in <module>
    aedtapp = Hfss(specified_version="2021.2", projectname=project_name, designname=design_name, non_graphical=non_graphical, new_desktop_session=False)
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\generic\design_types.py", line 66, in Hfss
    return app(
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\hfss.py", line 166, in __init__
    FieldAnalysis3D.__init__(
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\application\Analysis3D.py", line 78, in __init__
    Analysis.__init__(
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\application\Analysis.py", line 110, in __init__
    Design.__init__(
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\application\Design.py", line 194, in __init__
    self._desktop_class = Desktop(
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\desktop.py", line 569, in __init__
    self._init_desktop(non_graphical)
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\desktop.py", line 698, in _init_desktop
    if non_graphical or self._main.oDesktop.GetIsNonGraphical():
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\win32com\client\dynamic.py", line 638, in __getattr__
    raise AttributeError("%s.%s" % (self._username_, attr))
AttributeError: <unknown>.GetIsNonGraphical

Process finished with exit code 1

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

C:\Users\arobinso>python -m pip freeze
ansys-api-fluent==0.3.11
ansys-api-mapdl==0.5.1
ansys-api-meshing-prime==0.1.1
ansys-api-platform-instancemanagement==1.0.0b3
ansys-dpf-core==0.6.0
ansys-dpf-gate==0.2.1
ansys-dpf-gatebin==0.3.1
ansys-dpf-post==0.2.5
ansys-fluent-core==0.11.0
ansys-grantami-bomanalytics==1.0.1
ansys-grantami-bomanalytics-openapi==1.0.0
ansys-grpc-dpf==0.7.1
ansys-mapdl-core==0.63.2
ansys-mapdl-reader==0.52.11
ansys-meshing-prime==0.2.0
ansys-openapi-common==1.1.1
ansys-platform-instancemanagement==1.0.2
ansys-seascape==0.2.0
appdirs==1.4.4
cachetools==5.3.0
certifi==2022.12.7
cffi==1.15.1
charset-normalizer==3.1.0
clr-loader==0.2.5
colorama==0.4.6
contourpy==1.0.7
cryptography==40.0.1
cycler==0.11.0
fonttools==4.39.3
geomdl==5.3.1
google-api-core==2.10.1
google-api-python-client==2.84.0
google-auth==2.17.2
google-auth-httplib2==0.1.0
googleapis-common-protos==1.56.4
grpcio==1.53.0
h5py==3.8.0
httplib2==0.22.0
idna==3.4
imageio==2.27.0
importlib-metadata==6.1.0
kiwisolver==1.4.4
matplotlib==3.7.1
numpy==1.24.2
packaging==23.0
pandas==2.0.0
Pillow==9.5.0
platformdirs==3.2.0
plumbum==1.8.1
pooch==1.7.0
protobuf==3.20.1
protoc-gen-swagger==0.1.0
psutil==5.9.4
pyaedt==0.6.3
pyansys==2023.1.3
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycparser==2.21
pyiges==0.2.1
pyparsing==3.0.9
pypiwin32==223
pyspnego==0.8.0
python-dateutil==2.8.2
pythonnet==3.0.0rc6
pytwin==0.2.0
pytz==2023.3
pyvista==0.38.5
pywin32==306
requests==2.28.2
requests-negotiate-sspi==0.5.2
requests-ntlm==1.2.0
rpyc==5.0.1
rsa==4.9
scipy==1.10.1
scooby==0.7.1
six==1.16.0
tqdm==4.65.0
tzdata==2023.3
uritemplate==4.1.1
urllib3==1.26.15
vtk==9.2.6
zipp==3.15.0

svandenb-dev, 2023-06-28:
Hi @arobinson905,
is it specific only for 2021.2 ?

svandenb-dev, 2023-06-28:
@arobinson905 seems you are running an old Pyaedt version. Could you please upgrade ? 
from terminal 
pip install pyaedt -upgrade

arobinson905, 2023-06-28:
Yes, the instance works fine for 2022.2 and 2023.1, I didn’t try early versions than 2021.2

[External Sender]

maxcapodi78, 2023-06-29:
[image]
@arobinson905  as mentioned by @svandenb-dev maybe you need to update to latest pyaedt version. I've tested with latest version from main and it worked for me as you can see in the screenshot

arobinson905, 2023-06-29:
project_name = "C:\\Users\\arobinso\\OneDrive - ANSYS, Inc\\Documents\\CUBoulder\\Bowtie_T.aedt"
design_name = "Skin"
non_graphical = os.getenv("PYAEDT_NON_GRAPHICAL", "False").lower() in ("true", "1", "t")
aedtapp = Hfss(specified_version="2021.2", projectname=project_name, designname=design_name, non_graphical=non_graphical, new_desktop_session=False)

**I upgraded my pyaedt, but I am still getting the same error**

Traceback (most recent call last):
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\CUBoulder\pyaedt_script.py", line 11, in <module>
    aedtapp = Hfss(specified_version="2021.2", projectname=project_name, designname=design_name, non_graphical=non_graphical, new_desktop_session=False)
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\generic\design_types.py", line 66, in Hfss
    return app(
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\hfss.py", line 166, in __init__
    FieldAnalysis3D.__init__(
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\application\Analysis3D.py", line 78, in __init__
    Analysis.__init__(
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\application\Analysis.py", line 110, in __init__
    Design.__init__(
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\application\Design.py", line 194, in __init__
    self._desktop_class = Desktop(
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\desktop.py", line 569, in __init__
    self._init_desktop(non_graphical)
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\pyaedt\desktop.py", line 698, in _init_desktop
    if non_graphical or self._main.oDesktop.GetIsNonGraphical():
  File "C:\Users\arobinso\OneDrive - ANSYS, Inc\Documents\Ball PYP\Python Tutorial\CUBoulder\lib\site-packages\win32com\client\dynamic.py", line 638, in __getattr__
    raise AttributeError("%s.%s" % (self._username_, attr))
AttributeError: <unknown>.GetIsNonGraphical

Process finished with exit code 1

Samuelopez-ansys, 2023-06-29:
Hi @arobinson905 ,

Is there any reason to don't use 2022R2 ? As mentioned in the documentation, PyAEDT is compatible from 2022R2 and later:
https://aedt.docs.pyansys.com/version/stable/#dependencies

If there is any specific reason, we can try to debug the issue.

arobinson905, 2023-06-29:
Thanks Sam!

A customer wanted to use 2021.2, my mistake, I thought we supported HFSS 2021. I will tell them to upgrade

closed 2023-07-05 by Samuelopez-ansys