ansys/pyaedt #3248: Improve ...

nnetake, 2023-07-11:
### Description of the current limitations

Currently if we update the properties of the setup object we need to explicitly update the setup to set these properties. Updating setup explicitly after setting the property is not intuitive, may be setup.update() should be included in the set property.

        setup_obj = ipk.get_setup("Setup1")
        assert setup_obj is not None
        setup_obj.props["Convergence Criteria - Max Iterations"] = 499
        setup_obj.update()

the last line should not be needed.
I wasted some time in figuring this out.

### Code sample expected

        setup_obj = ipk.get_setup("Setup1")
        assert setup_obj is not None
        setup_obj.props["Convergence Criteria - Max Iterations"] = 499

Samuelopez-ansys, 2023-07-14:
Hi @nnetake ,

I don't know which PyAEDT version you are using, but I have tested the latest main and it works as expected:

[image]

nnetake, 2023-07-14:
Hi @Samuelopez-ansys thanks for looking into it. In the above steps, please create setup using pyaedt and get the setup object for this setup and then try to update iterations. Please let me know if this replicates the issue or else we can have a quick call.

I am using the latest dev version.

Samuelopez-ansys, 2023-07-17:
Hi @nnetake ,

It is working too:
[image]

I know the problem you have, you are first creating the setup with PyAEDT, and then using the method get_setup. The problem with this method is, it is getting the setup information from the .aedt file, and if you don't save the project and reload the pyaedt object it will not work. This method is useful when you open an existing project, then my suggestion is directly use the object returned by the create_setup, if you still need to get_setup then:

```
import pyaedt

pyaedt.settings.use_grpc_api = False
app = pyaedt.Icepak(specified_version="2023.1")

setup = app.create_setup()

app.save_project()

app.release_desktop(False, False)

app = pyaedt.Icepak(specified_version="2023.1")

setup_obj = app.get_setup("MySetupAuto")
setup_obj.props["Convergence Criteria - Max Iterations"] = 490

app.release_desktop(False, False)
```

In the future, we will remove the .aedt dependencies but the setup still needs this information, and we need to wait for some AEDT API new commands.

nnetake, 2023-07-17:
Okay, I understand it. Thanks for the information and your efforts, really appreciate the efforts of pyaedt development team.

closed 2023-07-17 by nnetake