ansys/pyaedt #3275: It is not possible to create two or more matched differential pairs

Jessi0396, 2023-07-14:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Please make it possible to create two or more matched differential pairs.
If you try to create more than one matched differential pair only one of them will be machted.

### Steps To Reproduce

- create more than one differential pair (in HFSS) with pyaedt
- set all of them matched=True
- You will see that only one of them is matched in Ansys

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

from tkinter import *
import tkinter as tk
from tkinter import ttk
from tkinter import messagebox
import os
import pyaedt
import math

dcrawforAtAnsys, 2023-08-07:
@Jessi0396: could you please add more information about the method/class where you see a defect?

dcrawforAtAnsys, 2023-08-12:
@Jessi0396 I am able to reproduce the issue. For future reference note the changes I made to your previous comment. Use of [markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) makes it much easier to understand the issue. I'll look into it and try to get this addressed ASAP.

dcrawforAtAnsys, 2023-08-14:
Issue is resolved with PR #3441.

closed 2023-08-14 by dcrawforAtAnsys

Jessi0396, 2023-08-15:
I tried to create two matched differential pairs. Ansys creates two differential pairs but there is still only one **matched** differential pair.