ansys/pyaedt #3316: AEDT Optimetrics Native Iron Python APIs

tizianrot, 2023-07-24:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Dear All,
not sure if it is the cprrect place but I would raise awareness and maybe you can suggest a workaround. 
The Native Iron Python API (Exporting results from an Optimetrics Setup in a Maxwell Design):
oModule.ExportDOEResponseSurface
works well if  called from within the AEDT GUI. If called from a python IDE through pyaedt, calling the oModule objecvt from a Maxwell 2d instance, throws a pywintypes_com_error as attached.
Thank you for any help.
[image]

 

### Steps To Reproduce

- set up a Maxwell 2d design
- set up an Optimetrics DOE 
- export responce surface results to a csv file

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

-

Samuelopez-ansys, 2023-08-24:
@tizianrot  Please could you share an end to end script to reproduce the issue?

Thanks

gmalinve, 2023-10-23:
Closing as agreed with @tizianrot. 
Issues fixed from v2023.2 by dev.

closed 2023-10-23 by gmalinve