ansys/pyaedt #3330: Improve pyaedt.circuit.Circuit() will create a new design if I do not assign a design name when open an existing project with multiple designs

JusMave, 2023-07-26:
### Description of the current limitations

### Description 
pyaedt.circuit.Circuit() will create a new design if I don't assign a design name when opening an existing project with multiple designs.

I hope it can always choose the top design rather than creating a new one.

### Log:
**PyAEDT INFO: Project xxxx set to active.
PyAEDT INFO: No consistent unique design is present. Inserting a new design.**

### Code sample expected

Circuit(projectname=circuitProjectPath, designname=None, non_graphical=False, new_desktop_session=False)

Samuelopez-ansys, 2023-07-27:
Hi @JusMave ,

This is expected behavior.

If in your project you have more than 1 Circuit design and you do not specify the design name, PyAEDT will create a new design.

It is in the PyAEDT core and all PyAEDT applications work like that, if the user does not pass the design name, and there is more than 1 design of the same type, PyAEDT can not decide which one is the user referring to. For more robust workflows, if you have more than 1 design, I suggest using the Desktop class, and you will decide to which design you want to connect.

In summary, this can not be changed because it was a core decision and many users will be affected.

Please let us know your feedback.

JusMave, 2023-07-27:
Thank you for your reply! I checked the Desktop class, but it seems it can't open a project through a project file path... Is there an approach to open a project without a design name and don't create a new design?

Samuelopez-ansys, 2023-07-27:
Hi @JusMave ,

We are just adding a new method in Desktop to do it, but in the meanwhile, this is the current approach I use:

1) Launch AEDT, it will open an empty AEDT session.
```
desktop = pyaedt.Desktop()
```

2) Open the project using AEDT API.

`desktop.odesktop.OpenProject(project_name)`

3) Get project info:
```
active_project = desktop.odesktop.GetActiveProject()
active_design = active_project.GetActiveDesign()
app_name = active_design.GetDesignType()
....
```

With this information, you should be able to connect to the design you want using the desired pyaedt application (Circuit, Hfss...). For example, if you could get the Circuit design name (circuit_name), then you need to call the application object pyaedt.Circuit(projectname=project_name, design_name=circuit_name).

JusMave, 2023-07-27:
I will try that, thank you so much!

closed 2023-08-07 by Samuelopez-ansys