ansys/pyaedt #3351: Allo to disable logger.info in method change_text_property

amichel0205, 2023-08-01:
### Description of the feature

In a script for creating a schematic using pyaedt, I have to create and modify like about 100 to 300 text. Method "change_text_property" is displaying 4 different message for each text modification, using logger.info,  in pyaedt.log, resulting in 1000+ lines in pyaedt.log(maybe slow down the execution of the script aas well).

I suggest to add an optional boolean parameter loggerinfo to the method.
Default is true, setting this parameter to False will disable logger.info for the method

### Steps for implementing the feature

def change_text_property(self, property_id, property_name, property_value, loggerinfo=True):
...
...
        if loggerinfo:
            self.logger.info("Property {} Changed correctly.".format(property_name))
        return True

amichel0205, 2023-08-02:
After discussing with Massimo we agreed that the best is to change self.logger.info("Property {} Changed correctly.".format(property_name)) with self.logger.debug("Property {} Changed correctly.".format(property_name)) as this information are mainly valuable for debugging.

gmalinve, 2023-10-20:
Issued tackled in PR #3362 thus closing issue.

Thanks @amichel0205

closed 2023-10-20 by gmalinve