ansys/pyaedt #3356: Icepak.create_pcb_from_3dlayout(component_name, project_name, design_name, extent_type='Polygon') still creates bounding box

Rungenwagen, 2023-08-01:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The function Icepak.create_pcb_from_3dlayout(component_name, project_name, design_name, extent_type='Polygon') doesn't  create a polygonal PCB successfully. Instead, AEDT uses bounding box. The AEDT window shows the following error.

[image]

It seems that it works only if I give the correct name of the outline polygon in the parameter of this function.
When I create a PCB manually, the outline polygon is automatically selected. But in scripting, I can't know the outline in priori. In a compatible script, I can use 
```python
oEditor.InsertNativeComponent(
	[
		"NAME:InsertNativeComponentData",
		"TargetCS:="		, "Global",
		"SubmodelDefinitionName:=", "PCB",
		......

		[
			"NAME:NativeComponentDefinitionProvider",
			"Type:="		, "PCB",
			"Unit:="		, "mm",
			"Version:="		, 0,
			"MovePlane:="		, "XY",
			"Use3DLayoutExtents:="	, True 
		]
               ......
	])
``` 
to avoid this problem. In PyAEDT there is no such parameter.

### Steps To Reproduce

1. Open AEDT and import an EDB file
2. Insert Icepak design
3. Run the script below.
```python
from pyaedt import Icepak

ipk = Icepak()
project_name = ipk.project_name

oEditor = ipk.modeler.oeditor
ipk.create_pcb_from_3dlayout(
    component_name='PCB',
    project_name=project_name,
    design_name='main',
    extent_type='Polygon'
)

ipk.release_desktop(close_projects=False, close_desktop=False)
``` 

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

attrs==23.1.0
certifi==2023.5.7
cffi==1.15.1
charset-normalizer==3.2.0
click==8.1.5
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.5
colorama==0.4.6
cycler==0.11.0
et-xmlfile==1.1.0
Fiona==1.9.4.post1
fonttools==4.38.0
geopandas==0.10.2
idna==3.4
imageio==2.31.1
importlib-metadata==6.7.0
kiwisolver==1.4.4
matplotlib==3.5.3
networkx==2.6.3
numpy==1.21.6
openpyxl==3.1.2
osmnx==1.1.2
packaging==23.1
pandas==1.3.5
Pillow==9.5.0
platformdirs==3.9.1
plumbum==1.8.2
pooch==1.7.0
psutil==5.9.5
pyaedt==0.6.85
pycparser==2.21
pyparsing==3.1.0
pyproj==3.2.1
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2023.3
pyvista==0.38.0
pywin32==306
requests==2.31.0
rpyc==5.3.1
Rtree==1.0.1
scikit-rf==0.28.0
scipy==1.7.3
scooby==0.7.2
Shapely==1.8.5.post1
six==1.16.0
SRTM.py==0.3.7
typing_extensions==4.7.1
urllib3==2.0.3
utm==0.7.0
vtk==9.2.6
zipp==3.15.0

KimMartinNielsen, 2023-08-02:
Hi Rungenwagen, 

I had the same issue - however I managed to overcome it by searching for the outline polygon the HFSS 3DL and use that input when importing PCB in Icepak. 
Using something like the lines below. 

""
Outline = h3d.modeler.objects_by_layer("Outline")
 
ipkapp.create_pcb_from_3dlayout(component_name="PCB1",project_name=h3d.project_name, design_name=h3d.design_name, extent_type='Polygon',outline_polygon=Outline[0])

""

I have not seen a project where the outline did not excist in HFSS 3DL - so I do not know what will happen in that case. 

Best regards,
Kim

Rungenwagen, 2023-08-07:
Hi Kim,

thank you for the tip. I managed it.

Best regards,
Rungenwagen

maxcapodi78, 2023-09-27:
Fixed, please have a  look

PR #3651 (2023-09-27): Fix bug in import PCB in Icepak to support Hfss3d Layout extent
Fix #3537 
Fix #3356
closed 2023-09-27 by maxcapodi78