ansys/pyaedt #3363: ANSYS Icepak2023R2 + PyAEDT

KimMartinNielsen, 2023-08-02:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Dear Team, 

Whenever I change the internal node power in a Thermal Network boundary condition, the code is changing and destroying the Links and other properties of that thermal network. 

I use the following code for changing the internal power: 

"""
ThermalB = ipk.boundaries
ThermalB[1].props["Nodes"]["Internal"]["Power"] = "10000mW"
ThermalB[1].update()
"""
Attached are two pictures of the thermal network, "before" and "after" running the code, changing thermal network of index =1. 
In addition to that, we see the following Error message from PyAEDT: 

"""

PyAEDT WARNING: Because the solution is steady state, neither mass nor specific heat assignment is relevant.
PyAEDT ERROR: KEY ERROR on _add_to_props
PyAEDT ERROR:   File "C:\Users\1990kini\AppData\Local\Programs\Python\Python310\lib\site-packages\pyaedt\modules\Boundary.py", line 4204, in add_link
PyAEDT ERROR:     self._add_to_props(new_link, "Links")
PyAEDT ERROR:   File "C:\Users\1990kini\AppData\Local\Programs\Python\Python310\lib\site-packages\pyaedt\modules\Boundary.py", line 3962, in _add_to_props
PyAEDT ERROR: Method arguments:
PyAEDT ERROR:     new_node = <pyaedt.modules.Boundary.NetworkObject._Link object at 0x0000024663B366B0>
PyAEDT ERROR:     type_dict = Links
PyAEDT ERROR: KEY ERROR on _add_to_props
PyAEDT ERROR:   File "C:\Users\1990kini\AppData\Local\Programs\Python\Python310\lib\site-packages\pyaedt\modules\Boundary.py", line 4204, in add_link
PyAEDT ERROR:     self._add_to_props(new_link, "Links")
PyAEDT ERROR:   File "C:\Users\1990kini\AppData\Local\Programs\Python\Python310\lib\site-packages\pyaedt\modules\Boundary.py", line 3962, in _add_to_props
PyAEDT ERROR: Method arguments:
PyAEDT ERROR:     new_node = <pyaedt.modules.Boundary.NetworkObject._Link object at 0x0000024663B36E30>
PyAEDT ERROR:     type_dict = Links
PyAEDT ERROR: KEY ERROR on _add_to_props

"""

I hope you can provide me with a solution or a fix. 

Thank you in advance. 

Best regards, 
Kim 
[image]

[image]

### Steps To Reproduce

Run following pyAEDT code on any icepak project containing one thermal network boundary condition. 

from pyaedt import Icepak

desktop = pyaedt.Desktop("2023.2", non_graphical=True)
project_name = ##Path to project 

ipkapp = Icepak(projectname=project_name, specified_version="2023.2")

ThermalB = ipkapp.boundaries

ThermalB[0].props["Nodes"]["Internal"]["Power"] = "10000mW"
ThermalB[0].update()

desktop.release_desktop()

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

aiohttp==3.8.1
aiosignal==1.2.0
altgraph==0.17.2
anyio==3.7.1
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
arrow==1.2.3
asttokens==2.2.1
async-lru==2.0.3
async-timeout==4.0.2
attrs==23.1.0
Babel==2.12.1
backcall==0.2.0
beautifulsoup4==4.12.2
bleach==6.0.0
cachetools==5.1.0
certifi==2022.5.18.1
cffi==1.15.1
charset-normalizer==2.0.12
clr-loader==0.2.3
colorama==0.4.6
comm==0.1.3
configobj==5.0.6
cycler==0.11.0
debugpy==1.6.7
decorator==5.1.1
defusedxml==0.7.1
et-xmlfile==1.1.0
exceptiongroup==1.1.2
executing==1.2.0
fastjsonschema==2.18.0
fonttools==4.33.3
fpdf==1.7.2
fqdn==1.5.1
frozenlist==1.3.1
future==0.18.2
google-api-core==2.8.0
google-api-python-client==2.48.0
google-auth==2.6.6
google-auth-httplib2==0.1.0
googleapis-common-protos==1.56.1
grpcio==1.46.1
httplib2==0.20.4
idna==3.3
imageio==2.22.0
ipycanvas==0.13.1
ipyevents==2.0.1
ipykernel==6.25.0
ipython==8.14.0
ipython-genutils==0.2.0
ipyvtklink==0.2.3
ipywidgets==7.7.5
isoduration==20.11.0
jedi==0.18.2
Jinja2==3.1.2
jproperties==2.1.1
json5==0.9.14
jsonpointer==2.4
jsonschema==4.18.4
jsonschema-specifications==2023.7.1
jupyter-events==0.6.3
jupyter-lsp==2.2.0
jupyter_client==8.3.0
jupyter_core==5.3.1
jupyter_server==2.7.0
jupyter_server_terminals==0.4.4
jupyterlab==4.0.3
jupyterlab-pygments==0.2.2
jupyterlab-widgets==1.1.4
jupyterlab_server==2.24.0
kaleido==0.2.1
kiwisolver==1.4.2
MarkupSafe==2.1.3
matplotlib==3.5.2
matplotlib-inline==0.1.6
mistune==3.0.1
multidict==6.0.2
nbclient==0.8.0
nbconvert==7.7.3
nbformat==5.9.1
nest-asyncio==1.5.6
notebook==7.0.0
notebook_shim==0.2.3
numpy==1.22.3
openpyxl==3.0.10
overrides==7.3.1
packaging==21.3
pandas==1.4.2
pandocfilters==1.5.0
parso==0.8.3
pefile==2022.5.30
pickleshare==0.7.5
Pillow==9.1.1
platformdirs==3.9.1
plotly==5.8.0
plumbum==1.7.2
prometheus-client==0.17.1
prompt-toolkit==3.0.39
protobuf==3.20.1
psutil==5.9.5
pure-eval==0.2.2
pyaedt==0.6.86
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycparser==2.21
Pygments==2.15.1
pyinstaller==5.3
pyinstaller-hooks-contrib==2022.8
pyparsing==3.0.9
python-dateutil==2.8.2
python-json-logger==2.0.7
pythonnet==3.0.1
pytz==2022.1
pyvista==0.36.1
pywin32==306
pywin32-ctypes==0.2.0
pywinpty==2.0.11
PyYAML==6.0.1
pyzmq==25.1.0
referencing==0.30.0
requests==2.31.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.9.2
rpyc==5.3.1
rsa==4.8
scooby==0.6.0
Send2Trash==1.8.2
six==1.16.0
sniffio==1.3.0
soupsieve==2.4.1
stack-data==0.6.2
tenacity==8.0.1
terminado==0.17.1
tinycss2==1.2.1
tk==0.1.0
tomli==2.0.1
tornado==6.3.2
traitlets==5.9.0
typing_extensions==4.7.1
uri-template==1.3.0
uritemplate==4.1.1
urllib3==1.26.9
vtk==9.2.0rc2
wcwidth==0.2.6
webcolors==1.13
webencodings==0.5.1
websocket-client==1.6.1
widgetsnbextension==3.6.4
wslink==1.8.2
XlsxWriter==3.0.3
yarl==1.8.1

closed 2023-08-11 by lorenzovecchietti

lorenzovecchietti, 2023-08-11:
Hi Kim, if you need to use pyAEDT with the fix straight away, please use this pip command to install pyAEDT from git:
`pip install pyaedt@git+https://github.com/ansys/pyaedt.git`

KimMartinNielsen, 2023-08-15:
Hi @lorenzovecchietti 

Thank you very much for your guidance. 

Best regards, 
Kim

KimMartinNielsen, 2023-08-23:
Hi @lorenzovecchietti 

After installing the pyeadt directly from GitHub, I have been trying out the changes in Thermal Network BC. But I am running into troubles when trying to access the Boundary object module. I am not sure if something has changed, or it is a bug? 
Please see error message below. 

[image]

Basically all I am trying to do is to access "ipkapp.boundaries" 

Test code is below: 

"""
desktop = pyaedt.Desktop("2023.2", non_graphical=True)
project_name = r"C:\RoadShowDemo\THA_Tool\TempFolder\Project_Fatemeh.aedt"

ipkapp = Icepak(projectname=project_name, specified_version="2023.2")
h3d = Hfss3dLayout(designname="PCB", specified_version="2023.2")

ThermalB = ipkapp.boundaries 

for i in range(len(ThermalB)):
    if ThermalB[i].name == "TN_U301":
        print("I am doing something")
        ThermalB[i].props["Nodes"]["Internal"]["Power"] = "10000mW"
        ThermalB[i].name = "KIM123_test"
        ThermalB[i].update()

ipkapp.save_project()   
"""

Also pip freeze output: 
[image]

I hope you are able to help? 

Best regards, 
Kim

lorenzovecchietti, 2023-08-23:
Can you please send me the file you are working on? I can't reproduce the error on my side

KimMartinNielsen, 2023-08-23:
Hi @lorenzovecchietti , 

I checked the project file, at the Thermal Network I was trying to modify was already broken. That caused the error. 
I am sorry. My mistake! 

Thank you again for the quick response. 

Best regards, 
Kim

PR #3393 (2023-08-11): Fix Schematic Data handling for Network Objects
Schematic Data were not parsed correctly (due to repeated keys that were being added to the design dictionary).

Close #3363