ansys/pyaedt #3386: Bug setting Solve Inside when material is a frequency dependent dielectric

adamgregory81, 2023-08-07:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When creating a 3D object in HFSS, the function that determines whether to set "Solve Inside" does not work when a Dielectric material uses a frequency dependent Bulk Conductivity.  If the string "Freq" is in the Bulk Conductivity expression, it always assumes it is a Conductor instead of a Dielectric

### Steps To Reproduce

Create a dielectric material with frequency dependence
Call a function from PYAEDT that makes a 3D object (like create_box) and assign the material to be a frequency dependent dielectric
Note that Solve Inside will be False

The Relative Permittivity may also need to be checked when Bulk Conductivity has a frequency dependent expression.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

aiohttp==3.8.3
aiosignal==1.2.0
altgraph==0.17.3
appdirs==1.4.4
async-timeout==4.0.2
attrs==22.1.0
cachetools==5.3.0
cffi==1.15.1
charset-normalizer==2.1.1
clr-loader==0.2.3
contourpy==1.0.5
cycler==0.11.0
et-xmlfile==1.1.0
fonttools==4.37.3
frozenlist==1.3.1
idna==3.4
imageio==2.22.0
kiwisolver==1.4.4
matplotlib==3.6.0
multidict==6.0.2
numpy==1.23.3
openpyxl==3.0.10
packaging==21.3
pandas==1.5.1
pefile==2023.2.7
Pillow==9.2.0
plumbum==1.7.2
psutil==5.9.2
py2exe==0.13.0.0
pyaedt==0.6.76
pycparser==2.21
pyinstaller==5.10.1
pyinstaller-hooks-contrib==2023.2
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.2.1
pyvista==0.36.1
pywin32==304
pywin32-ctypes==0.2.0
rpyc==5.3.1
scooby==0.6.0
six==1.16.0
vtk==9.1.0
wslink==1.8.2
yarl==1.8.1

Samuelopez-ansys, 2023-08-11:
Hi @adamgregory81 ,

Please share the snippet code to reproduce the issue.

Regards,
Samuel

dcrawforAtAnsys, 2023-08-19:
The main issues is in ``modeler.advanced_cad.stackup_3d.py`` in the method ``duplicate_parameterize_material()``.  Conductors are not properly identified.

This should be addressed in [this branch](https://github.com/ansys/pyaedt/tree/2977-bug-in-stackup3d).

dcrawforAtAnsys, 2023-09-14:
Should be resolved in PR #3566 .

closed 2023-09-14 by dcrawforAtAnsys