ansys/pyaedt #3425: create wave port does not work in the new Ansys Version (2023 R2)

Jessi0396, 2023-08-17:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The function "create wave port" in PyAEDT does not work in the new Ansys version (2023 R2). It worked in version 2022 R2.

q = pyaedt.Hfss(projectname=SpeicherungProjektname, designname="HFSSDesign1", solution_type = "Terminal", specified_version="2023.2", non_graphical=False, new_desktop_session=True)
id_port1 = q.modeler.get_faceid_from_position([0,position,0])
id_port2 = q.modeler.get_faceid_from_position([0,position,1])
#"position" is a variable that contains the position of the two ports on thy y-axis
port1 = q.create_wave_port(id_port1,None,None,portname="Port1",renorm=False,terminal_references="Mantel")
port2 = q.create_wave_port(id_port2,None,None,portname="Port2",renorm=False,terminal_references="Mantel")

### Steps To Reproduce

1. Get two or more faceids from position
2. Try to create a wave port with these faceids

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

from tkinter import *
import tkinter as tk
from tkinter import ttk
from tkinter import messagebox
import os
import pyaedt
import math

dcrawforAtAnsys, 2023-08-21:
The method `create_wave_port()` was deprecated with release 0.6.62.

If you're using the latest version of PyAedt, the following should work:
``` python
q = pyaedt.Hfss(projectname=SpeicherungProjektname, designname="HFSSDesign1", solution_type = "Terminal", specified_version="2023.2", non_graphical=False, new_desktop_session=True)
id_port1 = q.modeler.get_faceid_from_position([0,position,0])
id_port2 = q.modeler.get_faceid_from_position([0,position,1])
#"position" is a variable that contains the position of the two ports on thy y-axis
port1 = q.wave_port(id_port1, name="Port1", renorm=False, terminal_references="Mantel")
port2 = q.wave_port(id_port2, name="Port2", renorm=False, terminal_references="Mantel")
```

closed 2023-08-23 by dcrawforAtAnsys

Jessi0396, 2023-08-23:
Thank you.
Just a little correction after testing and searching for it. You have to write it like this to make it work:
port1 = q.wave_port(id_port1, reference="Mantel", name="Port1", renormalize=False)