ansys/pyaedt #3432: set marker in reports

Jessi0396, 2023-08-18:
### Description of the feature

Please mae it posssible to add marker in reports.

### Steps for implementing the feature

Choose report and the point where the marker should be.

gmalinve, 2023-08-21:
Hi @Jessi0396 ,

Have you already tried to use add_cartesian_x_marker() and add_cartesian_y_marker() for your purposes?
Please do let me know if this is what you intended to use and in case we can close the issue.

Giulia

Jessi0396, 2023-08-21:
Thank you for your answer.
I did not test it in the new (2023 R2) version of Ansys. I will try that!
Do you know where I find the description of this function on the PyAnsys website? I do not find it.

Samuelopez-ansys, 2023-08-21:
Hi @Jessi0396 ,

You can find it here:
https://aedt.docs.pyansys.com/version/stable/API/_autosummary/pyaedt.modules.report_templates.Standard.add_cartesian_x_marker.html#pyaedt.modules.report_templates.Standard.add_cartesian_x_marker

closed 2023-08-22 by Samuelopez-ansys

Samuelopez-ansys, 2023-08-24:
Hi @Jessi0396 ,

I reopened the issue. This method is not developped, but right now I do not have the time to implement it. You have some workaround if you need a solution now.

1) You can use the AEDT API command:

oModule = aedtapp.odesign.GetModule("ReportSetup")
oModule.AddMarker("Winding Plot 1", "m4", "InputCurrent(PhaseA) : BaseCase : Transient : Us=\'252.302067635866V\' : Cartesian", "240ms")

2) Take the data using get_solution_data, and use matplotlib to create the plot with the markers you need. Matplotlib has a lot usefull capabilities.

Jessi0396, 2023-08-25:
But that doesn't give me a solution in ansys itself, does it?

Samuelopez-ansys, 2023-08-25:
@Jessi0396 The first option, creates the marker in the HFSS specific trace. With PyAEDT you can combine PyAEDT commands and AEDT API commands like the 1st option I mentioned. You just need to record the script, and save in a variable the string of the trace you need, for example, "InputCurrent(PhaseA) : BaseCase : Transient : Us='252.302067635866V' : Cartesian"

From my point of view, I prefer the second option because once you get the solution data, you can do all post-processing in python, but if you prefer HFSS trace style, go ahead with the 1st option.

gmalinve, 2024-10-01:
Closing as duplicate of #4897.
Thanks for highlighting this @Jessi0396!
Hopefully someone will pick this up.