ansys/pyaedt #3433: Create ellipse creates rectangle instead

Jessi0396, 2023-08-18:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

ellipse1 = q.modeler.create_ellipse(cs_plane = "YZ", position = [0,1,0],major_radius = 1, ratio = 0.75, name = "Ellipse1", matname= "air")
with q = pyaedt.Hfss(projectname="Test", designname="HFSSDesign1", solution_type = "Terminal", specified_version="2023.2", non_graphical=False, new_desktop_session=True)
This function creates a rectangle instead of an ellipse.

### Steps To Reproduce

q = pyaedt.Hfss(projectname="Test" designname="HFSSDesign1", solution_type = "Terminal", specified_version="2023.2", non_graphical=False, new_desktop_session=True)
ellipse1 = q.modeler.create_ellipse(cs_plane = "YZ", position = [0,1,0],major_radius = 1, ratio = 0.75, name = "Ellipse1", matname= "air")
You will see that Ansys creates a rectangle.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

from tkinter import *
import tkinter as tk
from tkinter import ttk
from tkinter import messagebox
import os
import pyaedt
import math

gmalinve, 2023-08-18:
@Samuelopez-ansys 

If AEDT is run in grpc mode the issue does happen (for example add_material).
When we update the material in add_material and call the native API method AddMaterial, for names containing specific chars, the method returns None (where it should return the name of the material created).
If AEDT is run in COM mode, this doesn't happen. The native API method returns the material name.

For other methods using special chars (create_rectangle etc) if AEDT is run in COM mode the message manager displays an error saying that the name cannot contain non-ASCII chars.

gmalinve, 2023-08-21:
Hi @Jessi0396,

Thanks for submitting this issue.
create_ellipse() needed a small fix which should be fixed in #3440 .
Fix should be available in next release.

Giulia

closed 2023-08-21 by gmalinve