ansys/pyaedt #3446: Maxwell2D.post.get_solution_data

mehs1006, 2023-08-21:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

AEDT kept crashing when I used "Maxwell2D.post.get_solution_data".
Here I upload the error code in ".zip" file.
[Error_ExportData.zip](https://github.com/ansys/pyaedt/files/12398347/Error_ExportData.zip)

### Steps To Reproduce

`m2d = Maxwell2d(
    projectname=r"D:/Ansoft/MyProject.aedt", 
    designname="MyDesign",
    );`
`Variations = m2d.available_variations;`
`data = m2d.post.get_solution_data(
    expressions="Current(PhaseA)" , variations=Variations, 
    primary_sweep_variable="Time");`

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

alabaster==0.7.13
arrow==1.2.3
astroid==2.15.6
asttokens==2.2.1
atomicwrites==1.4.1
attrs==23.1.0
autopep8==2.0.2
Babel==2.12.1
backcall==0.2.0
bcrypt==4.0.1
beautifulsoup4==4.12.2
binaryornot==0.4.4
black==23.7.0
bleach==6.0.0
certifi==2023.7.22
cffi==1.15.1
chardet==5.1.0
charset-normalizer==3.2.0
click==8.1.6
cloudpickle==2.2.1
clr-loader==0.2.5
colorama==0.4.6
comm==0.1.3
contourpy==1.1.0
cookiecutter==2.2.3
cryptography==41.0.2
cycler==0.11.0
debugpy==1.6.7
decorator==5.1.1
defusedxml==0.7.1
diff-match-patch==20230430
dill==0.3.7
docstring-to-markdown==0.12
docutils==0.20.1
et-xmlfile==1.1.0
executing==1.2.0
fastjsonschema==2.18.0
flake8==6.0.0
fonttools==4.41.1
idna==3.4
imagesize==1.4.1
importlib-metadata==6.8.0
inflection==0.5.1
intervaltree==3.1.0
ipykernel==6.24.0
ipython==8.14.0
ipython-genutils==0.2.0
isort==5.12.0
jaraco.classes==3.3.0
jedi==0.18.2
jellyfish==1.0.0
Jinja2==3.1.2
jsonschema==4.18.4
jsonschema-specifications==2023.7.1
jupyter_client==8.3.0
jupyter_core==5.3.1
jupyterlab-pygments==0.2.2
keyring==24.2.0
kiwisolver==1.4.4
lazy-object-proxy==1.9.0
MarkupSafe==2.1.3
matlabengine==9.14.3
matplotlib==3.7.2
matplotlib-inline==0.1.6
mccabe==0.7.0
mistune==3.0.1
more-itertools==10.0.0
mypy-extensions==1.0.0
nbclient==0.8.0
nbconvert==7.7.2
nbformat==5.9.1
nest-asyncio==1.5.6
numpy==1.25.1
numpydoc==1.5.0
openpyxl==3.1.2
packaging==23.1
pandas==2.0.3
pandocfilters==1.5.0
paramiko==3.2.0
parso==0.8.3
pathspec==0.11.1
pexpect==4.8.0
pickleshare==0.7.5
Pillow==10.0.0
platformdirs==3.9.1
pluggy==1.2.0
plumbum==1.8.2
prompt-toolkit==3.0.39
psutil==5.9.5
ptyprocess==0.7.0
pure-eval==0.2.2
pyaedt==0.6.89
pycodestyle==2.10.0
pycparser==2.21
pydocstyle==6.3.0
pyflakes==3.0.1
Pygments==2.15.1
pylint==2.17.4
pylint-venv==3.0.2
pyls-spyder==0.4.0
PyNaCl==1.5.0
pyparsing==3.0.9
PyQt5==5.15.9
PyQt5-Qt5==5.15.2
PyQt5-sip==12.12.2
PyQtWebEngine==5.15.6
PyQtWebEngine-Qt5==5.15.2
python-dateutil==2.8.2
python-lsp-black==1.3.0
python-lsp-jsonrpc==1.0.0
python-lsp-server==1.7.4
python-slugify==8.0.1
pythonnet==3.0.1
pytoolconfig==1.2.5
pytz==2023.3
pywin32==306
pywin32-ctypes==0.2.2
PyYAML==6.0.1
pyzmq==25.1.0
QDarkStyle==3.1
qstylizer==0.2.2
QtAwesome==1.2.3
qtconsole==5.4.3
QtPy==2.3.1
referencing==0.30.0
requests==2.31.0
rope==1.9.0
rpds-py==0.9.2
rpyc==5.3.1
Rtree==1.0.1
scipy==1.11.1
six==1.16.0
snowballstemmer==2.2.0
sortedcontainers==2.4.0
soupsieve==2.4.1
Sphinx==7.1.0
sphinxcontrib-applehelp==1.0.4
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==2.0.1
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.5
spyder==5.4.3
spyder-kernels==2.4.4
stack-data==0.6.2
text-unidecode==1.3
textdistance==4.5.0
three-merge==0.1.1
tinycss2==1.2.1
tomli==2.0.1
tomlkit==0.11.8
tornado==6.3.2
traitlets==5.9.0
typing_extensions==4.7.1
tzdata==2023.3
ujson==5.8.0
urllib3==2.0.4
watchdog==3.0.0
wcwidth==0.2.6
webencodings==0.5.1
whatthepatch==1.0.5
wrapt==1.15.0
yapf==0.40.1
zipp==3.16.2

Samuelopez-ansys, 2023-08-22:
@mehs1006 Please share a dummy project to reproduce the issue.

mehs1006, 2023-08-22:
Hi @Samuelopez-ansys ,

I attached a dummy project in .zip file.
[DummyProject.zip](https://github.com/ansys/pyaedt/files/12411151/DummyProject.zip)

Thanks!

Samuelopez-ansys, 2023-08-24:
Hi @mehs1006,

First, the model you shared contains any confidential information? If yes, please reach out (samuel.lopez@ansys.com)

There are 2 problems in your script, the first one is related to the trace you want to get: Current(PhaseA)
As you can see in the screenshot, it does not exist in your setup (or I didn't find it), I only found InputCurrent(PhaseA):
[image]

The 2nd issue is related to the 'variations' parameter, you are passing the object, Variables, but as in the documentation of this method explains, you must pass a dictionary.

https://aedt.docs.pyansys.com/version/stable/API/_autosummary/pyaedt.modules.AdvancedPostProcessing.PostProcessor.get_solution_data.html

Then with this command you should get the information, and for example plot it using matplotlib:

```
data = m2d.post.get_solution_data(expressions="InputCurrent(PhaseA)", variations=Variations.all, primary_sweep_variable="Time")
data.plot()
```

[image]

Let me know if you have any question.

mehs1006, 2023-08-24:
Hi,

Thank you~ My problem is solved.
I didn't notice that "available_variations" becomes another type of data.

The detailed design of this IM model has been published in [https://doi.org/10.1109/ELMA52514.2021.9503055](url) and [https://doi.org/10.1109/VPPC53923.2021.9699149](url).
The material I used is also quite common, so I guess it is fine to leave it like this.

mehs1006, 2023-08-24:
Hi @Samuelopez-ansys ,
Thank you~ My problem is solved.
I didn't notice that "available_variations" becomes another type of data.

The detailed design of this IM model has been published in [https://doi.org/10.1109/ELMA52514.2021.9503055](url) and [https://doi.org/10.1109/VPPC53923.2021.9699149](url).
The material I used is also quite common, so I guess it is fine to leave it like this.

closed 2023-08-24 by mehs1006