ansys/pyaedt #3447: [Maxwell2D] skew model

mehs1006, 2023-08-21:
### Discussed in https://github.com/ansys/pyaedt/discussions/3434

<div type='discussions-op-text'>

<sup>Originally posted by **mehs1006** August 18, 2023</sup>
Hi,

In the Maxwell2D module, there's a function called "skew model".
I wonder if there's a command in PyAEDT that can make a skew model.
Thanks.
</div>

Samuelopez-ansys, 2023-08-22:
@gmalinve

gmalinve, 2023-08-22:
Hi @mehs1006 ,

Thanks for your request.
We don't have it yet but it's on our timeline.
We'll let you know as soon as is available.

Giulia

gmalinve, 2023-08-24:
@MaxJPRey

closed 2023-09-01 by maxcapodi78

gmalinve, 2023-09-01:
Hi @mehs1006 and @MaxJPRey ,

The skew model option in Maxwell2D has been implemented and will be available in the next release.
Hope this helps!

Giulia

MaxJPRey, 2023-09-01:
Thanks @gmalinve .

mehs1006, 2023-09-01:
Thanks @gmalinve